SCADA-LTS/Scada-LTS #2518: Create Data/Point Source with API

odenisnobre, 2023-04-19:
I have to create 10 points source(Modbus TCP) where what changes is only the end of the ip xxx.xxx.xxx.110@119

And the points are the same records and names for each point source.

Would it be possible to create these points sources and data points via API or in some automated way?

Limraj, 2023-04-19:
Hi @odenisnobre,
Login 
`http://localhost:8080/ScadaBR/api/auth/admin/admin GET`
Create new datasource:
`http://localhost:8080/ScadaBR/api/datasource POST`
example body:
```
{
    "xid": "DS_7973799",
    "name": "modbus_ip_1",
    "type": 3,
    "enabled": true,
    "connectionDescription": "common.default",
    "updatePeriodType": 1,
    "updatePeriods": 5,
    "quantize": false,
    "timeout": 500,
    "retries": 2,
    "contiguousBatches": false,
    "createSlaveMonitorPoints": false,
    "maxReadBitCount": 2000,
    "maxReadRegisterCount": 125,
    "maxWriteRegisterCount": 120,
    "transportType": "TCP",
    "host": "localhost",
    "port": 502,
    "encapsulated": false,
    "createSocketMonitorPort": false
}
```
Update datasource:
`http://localhost:8080/ScadaBR/api/datasource PUT`
body:
```
{
    "id": 1,
    "xid": "DS_7973799",
    "name": "modbus_ip_1",
    "type": 3,
    "enabled": true,
    "connectionDescription": "common.default",
    "updatePeriodType": 1,
    "updatePeriods": 5,
    "quantize": false,
    "timeout": 500,
    "retries": 2,
    "contiguousBatches": false,
    "createSlaveMonitorPoints": false,
    "maxReadBitCount": 2000,
    "maxReadRegisterCount": 125,
    "maxWriteRegisterCount": 120,
    "transportType": "TCP",
    "host": "localhost",
    "port": 502,
    "encapsulated": false,
    "createSocketMonitorPort": false
}
```
Create datapoint:
`http://localhost:8080/ScadaBR/api/datapoint POST`
body:
```
{
    "xid": "DP_6352",
    "name": "dp_2",
    "description": null,
    "enabled": true,
    "dataSourceTypeId": 3,
    "dataSourceId": 5,
    "deviceName": "modbus_ip_1",
    "pointLocator": {
        "dataSourceTypeId": "3",
        "dataTypeId": 2,
        "settable": true,
        "range": 3,
        "modbusDataType": 2,
        "slaveId": 1,
        "slaveMonitor": false,
        "socketMonitor": false,
        "offset": 0,
        "bit": 0,
        "registerCount": 0,
        "charset": "ASCII",
        "settableOverride": true,
        "multiplier": 1.0,
        "additive": 0.0,
        "relinquishable": false
    },
    "datasourceName": "modbus_ip_1",
    "dataSourceXid": "DS_797379",
    "typeId": 1,
    "settable": true
}
```
range:
Coil status - 1
Input status - 2
Holding register - 3
Input register - 4

Update datapoint:
`http://localhost:8080/ScadaBR/api/datapoint PUT`
body:
```
{
    "id": 12,
    "xid": "DP_6352",
    "name": "dp_2",
    "description": null,
    "enabled": true,
    "dataSourceTypeId": 3,
    "dataSourceId": 5,
    "deviceName": "modbus_ip_1",
    "pointLocator": {
        "dataSourceTypeId": "3",
        "dataTypeId": 2,
        "settable": true,
        "range": 3,
        "modbusDataType": 2,
        "slaveId": 1,
        "slaveMonitor": false,
        "socketMonitor": false,
        "offset": 0,
        "bit": 0,
        "registerCount": 0,
        "charset": "ASCII",
        "settableOverride": true,
        "multiplier": 1.0,
        "additive": 0.0,
        "relinquishable": false
    },
    "datasourceName": "modbus_ip_1",
    "dataSourceXid": "DS_797379",
    "typeId": 1,
    "settable": true
}
```

Regards,
Kamil Jarmusik

odenisnobre, 2023-04-19:
Wow, sensational.
I'll try.

Limraj, 2023-04-20:
It is up to version 2.7.5.2

Limraj, 2023-04-20:
@odenisnobre 
I think your reaction is overreacting. I'm just asking what you meant by that, it could have been a mockery. I try to help as much as I can, but I have no influence on whether you do it according to the instructions or in your own way. Generally, if something does not work for you as expected, it is worth asking before drawing conclusions. Also take into account that my English is not the best and we communicate by text, so we can't read the intention. In such communication, a subdued form of expression is more effective.

odenisnobre, 2023-04-20:
@Limraj 

Well, I think you misunderstood me.
My English is not the best either.
But my reaction was one of thanks rather than mockery.
Regardless of the result and conclusions, just having the possibility and having already thought about this possibility is very important.
This was my reaction, of thanks.
But in the next posts they will post only one: 'ok', 'all right'. That makes it simpler.

Limraj, 2023-04-20:
@odenisnobre Okey. We don't have to, but that statement was a bit above the norm... It's just that it surprised me a little. I think everything is fine. Let me know if you managed to create these Data Sources and Data Points.

odenisnobre, 2023-04-20:
Alright, I'll improve communication in the next ones.

I tested it and I still wasn't successful, it seems to me that ScadaBR is not available. I'm trying to see what's going on.

Limraj, 2023-04-20:
@odenisnobre 
What version of Scada-LTS are you working on? Are you using postman?

Limraj, 2023-04-20:
Are you using postman to call the API

odenisnobre, 2023-04-20:
He doesn't, I use other resources.
But if so, I can use it.

odenisnobre, 2023-04-20:
[image]

Limraj, 2023-04-20:
If you have Scada-LTS.war deployed then instead of ScadaBR you do Scada-LTS.

odenisnobre, 2023-04-20:
It was right.
The datasource and datapoints were created via postman.
Thank you very much.
I'll try now with the tool I use based on nodejs
Thank you very much.

Limraj, 2023-04-25:
@odenisnobre 
Do you have any more questions in this regard?

closed 2023-05-08 by Limraj