SCADA-LTS/Scada-LTS #2546: Improvement of the alarms function

Limraj, 2023-06-01:
**Description**
Improvement of the alarms function:
- [ ] 1. The maximum alarm resets when it reaches "return to normal";
- [ ] 2. Added the 'Acknowledge', 'Silence' functions from 'Pending alarms' to the 'Event search' result table;
- [ ] 3. Ability to sort by any of the columns; (?)

**Screenshots**
[image]
[image]

**Spec:**
 - Version 2.7.5.3

fabiodurao, 2023-06-01:
If you can contribute inspiration, in ScadaBR 1.2 this has been improved and several options have been added in the search tool (which can be seen below), it would be nice to have that in Scada-LTS too.

[image]

Limraj, 2023-06-04:
Hi @fabiodurao,
You need to create an issue and describe exactly which functions are in question. 
I'm just asking you to carefully check which functions are already in Scada-LTS and which are actually missing.

Regards,
Kamil Jarmusik