SCADA-LTS/Scada-LTS #2588: do not emit sounds

GabrielTec251, 2023-06-29:
I have problems with the audio output of the alarms, the alarm does not sound.

when I configure the alarms, whether light or urgent or critical, when triggered they do not emit any sound, I tested it in other browsers and enabled the permission of sites, the error persisted.

this message was translated from portuguese to english by google translator.

SCADA - LTS - 2.7.5.2
SCADA - LTS - 2.7.5.3

Limraj, 2023-06-29:
Hi @GabrielTec251,
We haven't used this feature for a long time. Could you check if v2.6.18 has the same problem?

Regards,
Kamil Jarmusik

GabrielTec251, 2023-06-29:
YES, have the same problem

GabrielTec251, 2023-06-29:
in SCADABR we configured the alarms and the beep sound came from the system itself.

I tested this LTS function on windows 10 and 11, on edge and chrome.

GabrielTec251, 2023-06-29:
[image]

the files exist, it only plays the media.

GabrielTec251, 2023-06-29:
[image]

here is how to be in scadaBR

Limraj, 2023-06-29:
@GabrielTec251 We will fix this in version 2.7.5.4

fabiodurao, 2023-06-30:
To help in an additional context, ScadaBR has also updated soundmanager to the version of this site (https://www.schillmania.com/projects/soundmanager2/doc/download/), follow the version link (http://www. wcny.org/js/soundmanager/download/soundmanagerv297a-20120318.zip)

It looks like the old version of soundmanager depended on flash, which is blocked by current browsers, and the new version relies on html5.

Limraj, 2023-06-30:
Hi @fabiodurao,
this information was not necessary, I already did it yesterday. 
This was the main problem with this feature.

Regards,
Kamil Jarmusik

closed 2023-07-04 by Limraj

GabrielTec251, 2023-08-08:
It is emitting sounds, however The alarm function must act in a way to "disturb" the operator, that is, there are cases in which the operator is not a few meters away from the supervisor, or even talking to a co-worker, for example. That's why the alarm must keep beeping until the operator acknowledges the alarm.