SCADA-LTS/Scada-LTS #2595: The configurable checkbox inside the graphic component cannot be unchecked

fabiodurao, 2023-06-30:
**Describe the bug**
It is not possible to save the graphic component with the option "configurable" unchecked. This problem is only visible when the "display controls" option is checked.

**To Play**
Steps to reproduce the behavior:
1. Go to the graph screen and create an "analog graph"
2. Place the mouse over the component and access "Edit point component settings"
3. Check the "display controls" option and try to uncheck the configurable option and you will see that when saving the option is not maintained.

**Expected behavior**
It was expected that with the "display controls" option checked without the configurable option, only a popup with information to be read and without a writing option would be seen.

**Screenshots**
[image]

**Desktop (please complete the following information):**
  - OS: Windows 11 64 bit
  - Browser: Chrome 114.0.5735.199 (Official version) 64 bits
  - Version: SLTS 2.7.5.3

**Additional context**
Add any other context about the problem here.

Limraj, 2023-06-30:
This is set based on user permissions, if the point is settable, the user has set permission on the view and the Data Point/Data Source, then they can set it, I don't see any point in complicating it even more. We configure it through permissions per user.

fabiodurao, 2023-07-01:
Thanks for the feedback, in ScadaBR 1.2 the behavior is different, the user may not present the user configurable option, even if he has permission, but if this is something intentional, I believe we can close the topic.

Limraj, 2023-07-01:
From ScadaBR 1.0 we inherited permissions not working properly, so check if permissions for user other than admin in ScadaBR 1.2 are working. If you report something and want it to be prioritized, then carefully check the operation on ScadaBR 1.2, do not speak from conviction, just test it on both applications. Beliefs are deceptive, and many times when you reported something, it turned out to work the same on ScadaBR 1.2.

closed 2023-07-01 by Limraj