SCADA-LTS/Scada-LTS #2596: The selection of images in the binary graphic component is not shown to the user

fabiodurao, 2023-06-30:
**Describe the bug**
It is not possible to see which images were selected when creating a binary graphic component.

**To Play**
Steps to reproduce the behavior:
1. Go to the graph screen and create a "binary graph"
2. Place the mouse over the component and access "Edit graphical renderer".
3. In "Image Set" select "Leds24" and select one image for state zero and another image for state one. A border will not appear around selected images.

**Expected behavior**
It was expected that when selecting the images for each state, there would be a visual feedback showing the user if the images were selected correctly.

**Screenshots**
[image]

**Desktop (please complete the following information):**
  - OS: Windows 11 64 bit
  - Browser: Chrome 114.0.5735.199 (Official version) 64 bits
  - Version: SLTS 2.7.5.3

fabiodurao, 2023-07-04:
I remembered that there was no context. In ScadaBR this appeared as a square border around the image, I assume this setting was lost in the css when they had changes to the STLS style. I put this information too, because it was considered a feature request, I believe it would be a bug fix.

Limraj, 2023-07-05:
@fabiodurao 
I suppose it wasn't in the ScadaBR 1.0 version we were basing.

fabiodurao, 2023-07-05:
I don't know if I understood what you said correctly, but from what I remember, Scada-LTS is a branch of ScadaBR 1.1, this version of SBR 1.1 was developed around 2016 for CASAN (water and sewage company in the state of santa Catarina in Brazil).

Limraj, 2023-07-22:
https://github.com/SCADA-LTS/Scada-LTS/issues/2520

closed 2023-08-17 by Limraj