SCADA-LTS/Scada-LTS #2597: Component "link" takes the absolute address of a page, which does not work with remote accesses

fabiodurao, 2023-06-30:
**Describe the bug**
Accessing the links does not work when accessing from computers other than the server, because the link shows the absolute value of the link (http://localhost:8080/Scada-LTS/views.shtm?viewId=2).

**To Play**
Steps to reproduce the behavior:
1. Go to the graphics screen and create a "link" component
2. Place the mouse over the component and access "Edit static content".
3. Select one of the "Views" to create the link to access it and in the "Link (href)" field the complete address will appear (http://localhost:8080/Scada-LTS/views.shtm?viewId=2) , and the best thing would be to show the relative address (views.shtm?viewId=2), which would allow the link to work for anyone accessing Scada-LTS from other computers.

**Expected behavior**
It was expected that after saving the screen, when clicking on the link, the page would lead to the link, which only happened if we were accessing it from the server.

**Screenshots**
[image]

**Desktop (please complete the following information):**
  - OS: Windows 11 64 bit
  - Browser: Chrome 114.0.5735.199 (Official version) 64 bits
  - Version: SLTS 2.7.5.3

Limraj, 2023-06-30:
Hi @fabiodurao,
Why don't you consult these topics? You think that if the address is absolute, then from another computer this link cannot be opened? After all, in this context, it doesn't matter if the link is absolute or relative... If the client computer has access to the application, it doesn't matter what address it has. Absolute or relative address leads to the same location of the server.

The situation is different from the configuration side, it's about portability of configuration between different servers (import/export), here we can see the advantages of relative addresses. 

Regards,
Kamil Jarmusik

fabiodurao, 2023-07-01:
We can discuss the matter further, but I believe that in this case there is no need and maybe I was not clear in the description of the problem.

Scenario:
2 computers
- Server (Scada-LTS installed): 192.168.1.100
- PC (client access to Scada-LTS)

If during the configuration of the link component the url is kept as "http://localhost:8080/Scada-LTS/views.shtm?viewId=2", when the PC accesses the application using the address 192.168.1.100:8080/Scada -LTS and click on the link, the PC browser will look for a page inside his computer (localhost) and will not open the desired address, result, error 404.

If the component's url is changed to "views.shtm?viewId=2", when the PC accesses the application and clicks on the link, its browser will automatically fill in the application address and complete it with the link address 192.168.1.100: 8080/Scada-LTS/views.shtm?viewId=2.

Please test and let me know if you have a different result.

Limraj, 2023-07-01:
Hi @fabiodurao,
So you are saying that you connect from your computer to the Scada-LTS application that is hosted on this server, you configure the links and then you get localhost there(?), so the link does not work for you and other computers? Do you think if it worked like that, could this feature be of any use that someone wouldn't have noticed before?

Again, this is a question of configuration portability, here from localhost, to another server. If as a client you connect to an application hosted on that server and set up links there, you get absolute links to that server, then as long as the client machine has access to that application on that server it must work. It doesn't matter if the link is relative or absolute.

There is also a situation where you configure it directly from the server, then you could get localhost there, the solution here is not to configure the application from the server, but from the client's computer, most clients use the application this way and have not reported this problem.

Regards,
Kamil Jarmusik

Limraj, 2023-07-01:
This topic can be closed, setting relative addresses will also solve the problem of configuring the application directly on the server.

closed 2023-07-01 by Limraj