SCADA-LTS/Scada-LTS #2606: MySQL with Scada-LTS ver 2.7.5.3  - Troubleshooting

AristotelesMelo, 2023-07-04:
**Is your feature request related to a problem? Please describe.**
I could not install Scada-LTS 2.7.5.3 on my windows 10 environment. I got the following message: "Access denied for user root@localhost using password :YES"

**Describe the solution you'd like**
I think the problem was with the access to the Mysql database, which the 2.7.5.3 installer itself created.

**Describe alternatives you've considered**
Considering that the installation of Scada-LTS 2.6.18 occurred without any problems, I suggest to remove the MySQL database installation from the next Scada-LTS installation version. 

Clarifying:  In the next version of Scada-LTS installer do not install MySQL together, exactly as in version 2.6.18.

Limraj, 2023-07-04:
Hi @AristotelesMelo,
Can you open cmd and run on the exe files you used to install Scada-LTS 2.6.18 and 2.7.5.3 the command:
`certutil -hashfile parth_to_file_exe MD5`
 and send me what this command returns?

Regards,
Kamil Jarmusik

AristotelesMelo, 2023-07-04:
Hi Kamil,

 

Thank you for your attention.

Where is the certutil file located?

 

BR,

Enviada em: terça-feira, 4 de julho de 2023 10:28
Para: SCADA-LTS/Scada-LTS ***@***.***>
Cc: AristotelesMelo ***@***.***>; Mention ***@***.***>

Limraj, 2023-07-04:
This is a command, you should have it available in windows 10. It's about executing this command on the exe files that you downloaded to install 2.6.18 and 2.7.5.3.

AristotelesMelo, 2023-07-04:
I have a download directory where all the scada-lts programs are stored.

 

When I run certutil, I get the following message:

 

D:\Downloads\scada\scada-lts>certutil -hashfile parth_to_file_exe MD5

CertUtil: -hashfile comando FALHOU: 0x80070002 (WIN32: 2 ERROR_FILE_NOT_FOUND)

CertUtil: O sistema não pode encontrar o arquivo especificado.

 

D:\Downloads\scada\scada-lts>

Enviada em: terça-feira, 4 de julho de 2023 10:40

Limraj, 2023-07-04:
You need to replace 'parth_to_file_exe'  with path to exe file for version 2.6.18 you used and same for exe file for version 2.7.5.3

AristotelesMelo, 2023-07-04:
D:\Downloads\scada\scada-lts>certutil -hashfile D:\Downloads\scada\scada-lts\ScadaLTS_v2.6.18_Setup.exe MD5

MD5 hash de D:\Downloads\scada\scada-lts\ScadaLTS_v2.6.18_Setup.exe:

a010198b35f1bcbaa248b452d2c58122

CertUtil: -hashfile : comando concluído com êxito.

 

D:\Downloads\scada\scada-lts>certutil -hashfile D:\Downloads\scada\scada-lts\ScadaLTS_v2.7.5.3_Beta_Standalone_Setup.exe MD5

MD5 hash de D:\Downloads\scada\scada-lts\ScadaLTS_v2.7.5.3_Beta_Standalone_Setup.exe:

a2e486b31601279654bedb5577d4d65f

CertUtil: -hashfile : comando concluído com êxito.

Enviada em: terça-feira, 4 de julho de 2023 11:06

AristotelesMelo, 2023-07-04:
D:\Downloads\scada\scada-lts>certutil -hashfile D:\Downloads\scada\scada-lts\ScadaLTS_v2.6.18_Setup.exe MD5 
MD5 hash de D:\Downloads\scada\scada-lts\ScadaLTS_v2.6.18_Setup.exe: 
a010198b35f1bcbaa248b452d2c58122 
CertUtil: -hashfile : comando concluído com êxito. 

 

D:\Downloads\scada\scada-lts>certutil -hashfile D:\Downloads\scada\scada-lts\ScadaLTS_v2.7.5.3_Beta_Standalone_Setup.exe MD5

MD5 hash de D:\Downloads\scada\scada-lts\ScadaLTS_v2.7.5.3_Beta_Standalone_Setup.exe: 
a2e486b31601279654bedb5577d4d65f 
CertUtil: -hashfile : comando concluído com êxito.

Limraj, 2023-07-04:
If you installed 2.6.18 using the installer without mysql server, i.e. you already have it installed on your system, then uninstall 2.7.5.3, then install using the old installer from 2.6.18, and to install the new version of Scada-LTS follow the steps:
1. Stop tomcat;
2. In "...Program Files/Scada-LTS/tomcat64/webapps" delete the Scada-LTS folder and the Scada-LTS.war file;
3. [From](https://github.com/SCADA-LTS/Scada-LTS/releases), download the interoperable version of the war file with Scada-LTS;
4. Copy the downloaded war file to "...Program Files/Scada-LTS/tomcat64/webapps";
5. Start tomcat;

**Just remember that between version v2.6.18 and v2.7.5.3 there is a big difference in the structure of the database, which means that the data after running version v2.7.5.3 will be incompatible with version v2.6.18, so going back to the v2.6.18 version without making a backup base will be very difficult.**

Before running the new version of Scada-LTS(on v2.6.18), make a database backup:
1. Open cmd:
2. Navigate to .../Program Files/Scada-LTS/mysql/bin:
`cd ".../Program Files/Scada-LTS/mysql/bin"`
3. Execute the command:
`mysqldump.exe --single-transaction --routines --triggers -u root -proot -P 3306 scadalts > scadalts-backup_2023_07.sql`
4. Copy the scadalts-backup_2023_07.sql to safe place;

Regards,
Kamil Jarmusik

Limraj, 2023-07-04:
@AristotelesMelo Even if you uninstalled version 2.6.18 before installing 2.7.5.3, then it should also be ok, only then you would have a clean database.

AristotelesMelo, 2023-07-04:
Kamil,

 

I would like to thank you for your attention and patience.

 

The version 2.7.5.3 is installed and OK.

 

Best Regards,

Enviada em: terça-feira, 4 de julho de 2023 11:57

AristotelesMelo, 2023-07-05:
Hi Kamil,

 

Please….

 

Yesterday after your help everything worked fine.

They were installed on Windows 10 ver 22H2 x64:

- jdk11 / DBeaver 23.1.1 / jdk 11 /MySQL 5.7 /Tomcat 9

 

I found it better to upgrade MySQL to version 8 and Scada-LTS stopped again.

I just uninstalled MySQL 5.7 and installed MySQL 8.0 with the same login and password on port 3306.

 

All tables have been created but it seems to me that the problem is when Scada-LTS tries to access MySQL.

 

The Tomcat is okay too.

 

But Scada-LTS not run

 

Is version 2.7.5.3 not compatible with MySQL 8.0?

I attach a log from Scada-LTS that I found important.

 

The database is working correctly:

 

Thanks in advance.

 

 

BR

AristotelesMelo, 2023-07-05:
Hi Kamil,

 

Analyzing the logs I added the missing fields to the events, reports and users tables and scada-lts is working again.

 

Do you think this installation is safe to proceed with development?

 

Tks

Limraj, 2023-07-05:
Hi @AristotelesMelo,
You're not providing enough information to be able to help. I wrote in the previous comment what is needed to be able to address your problem. Scada-LTS works with mysql server version 8.0, I suppose every version of Scada-LTS works with this server, because we didn't change the application to make it work with 8.0. 

Regards,
Kamil Jarmusik

AristotelesMelo, 2023-07-05:
System exception!

The server has experienced an exception processing your last request. The exception has been logged, and system administrators will be notified of this problem. You may continue to use the site. We apologize for the inconvenience.

Please provide the following code to the administrator to identify the problem:

20230705110124399

Limraj, 2023-07-05:
1. The first thing is to describe exactly the steps you took, how did you install the base and Scada-LTS, did you uninstall Scada-LTS before installing the new version. If not, just do it, uninstall 2.6.18 and you have no database anyway, and install 2.7.5.3, in the installer the database is available on port 3308, so there should be no conflicts with the database you have installed on your system;

2. Second, send screenshots or error logs so we can see what's going on.

3. Thirdly, you cannot add any fields/columns in the database yourself. If some columns are missing, it's most likely because the migration didn't go through, you can send what you have in the schema_version table, probably some step failed, which interrupted the migration and skipped some steps of the migration.

AristotelesMelo, 2023-07-05:
Okay. It´s running now.

More one time, thank you for your attention and support.

 

BR

Enviada em: quarta-feira, 5 de julho de 2023 11:17

Limraj, 2023-07-06:
Okay, so I'm closing.

closed 2023-07-06 by Limraj