SCADA-LTS/Scada-LTS #2608: A server error has occured. More information may be available in the console.

AristotelesMelo, 2023-07-06:
**Describe the bug**
I installed 2.7.5.3 (jdk-11/ Dbeaver23 / following the guidance presented on the site.

After installation Scada-LTS starts asking for login and password. So far so good

Any operation I try to do I get the message:

[image]
[image]

All navigation in scada-lts is compromised by showing these errors.

Limraj, 2023-07-06:
Hi @AristotelesMelo,
Session in tomcat lasts 30 minutes, after that you are automatically logged out of the application and then you get this message.
It's best to open a topic first at: https://github.com/SCADA-LTS/Scada-LTS/discussions
If it turns out that there is something to fix, then issue.

Regards,
Kamil Jarmusik

AristotelesMelo, 2023-07-06:
Hi Kamil

 

Thank you for your attention.

I didn't quite understand your answer.

These error messages appear constantly since the applications are started(tomcat + scada-lts). 

In the link you sent me I did not read any guidance to the problem I encountered.

Enviada em: quinta-feira, 6 de julho de 2023 11:07
Para: SCADA-LTS/Scada-LTS ***@***.***>
Cc: AristotelesMelo ***@***.***>; Mention ***@***.***>

Limraj, 2023-07-06:
I don't understand what constantly means, as I said, if you are not logged in to the application, you will get this message.
Describe exactly what you do when this error occurs.
In the future, in order to be able to help, you must try to describe, show the problem in such a way that the other person can reproduce it.

AristotelesMelo, 2023-07-06:
I – Login OK

 

II – Open scada-lts ok

 

III – Try save any alteration.

Enviada em: quinta-feira, 6 de julho de 2023 11:40

Limraj, 2023-07-06:
I'm sorry but I really don't know how to help you, you don't listen to instructions, the information you provide is not enough...

Limraj, 2023-07-07:
Hi @AristotelesMelo,
You don't know how to query the database, I thought you wrote that you were adding some fields/columns in the database, maybe I misunderstood you.If you don't know how to query the database:
1. You must have some client, in windows it can be [workbench -> select windows](https://dev.mysql.com/downloads/workbench/) , but I prefer [DBvear](https://github.com/SCADA-LTS/windows-installer/blob/master/README.md#12-client-mysql---dbeaver-optional---).
2. Then create a tutorial connection with:
https://github.com/SCADA-LTS/windows-installer/blob/master/README.md#12-client-mysql---dbeaver-optional--- . For server database from installer port: 3308, empty password:, user:root. If you have a workbench, the configuration is analogous, if you have a problem, let me know.
3. Create a new SQL query and execute this query based on it:
3.1. 'Set as default' for database scadalts:
[image]
3.2. Next on main menu, click arrow for SQL, and click 'New SQL script':
[image]
4. Send me logs from ".../Program Files/Scada-LTS/tomcat/logs" -> It's about logs catalina.out/mango.log/catalina.err which are non-empty.

Let me know if you have trouble following these steps. Most likely, the problems come from the fact that the migration failed, and then such errors can appear.

Regards,
Kamil Jarmusik

AristotelesMelo, 2023-07-10:
Hi Jamil

 

Once again I want to thank you for your attention.

After your help, version 2.75 is working.

 

As a suggestion, I think it would be interesting for the next versions to keep MySQL independent of Tomcat.

 

BR

Aristóteles

Enviada em: sexta-feira, 7 de julho de 2023 16:25

 

 <https://user-images.githubusercontent.com/35842300/251836403-1a1bec33-adc9-4e49-8a14-1bf0275bee57.png> 
 <https://user-images.githubusercontent.com/35842300/251836935-5ba0be4b-cf0d-4b7d-bb22-7ce0efe46ccd.png>

Limraj, 2023-07-10:
I'm closing this topic, if you have a problem, then try to do what I've already written to you, and if not, then continue on your own.

closed 2023-07-10 by Limraj