SCADA-LTS/Scada-LTS #2614: Duplicated scrollbar after a while

GabrielTec251, 2023-07-12:
the scrollbar after a period doubles, that is, I have two scrollbars on the screen.

**Desktop dell optiplex 3000:**
 - OS: win 11
 - Browser : chrome
 - Version 2.7.5.3

[image]

Limraj, 2023-07-12:
Hi @GabrielTec251,
Isn't some of your css/html/js causing this? 
There is no such error on the clean page?

Regards,
Kamil Jarmusik

GabrielTec251, 2023-07-12:
even without css it appears, I did the test on edge and it did not show this error
[image]

Limraj, 2023-07-12:
The question is does it appear when you give the background itself? If it doesn't appear then the css/html/js is the culprit.

GabrielTec251, 2023-07-12:
it does not appear when I paste the background, but this application also did not appear after placing the codes, it only appeared today

GabrielTec251, 2023-07-12:
SETPOINT

var s = "<input type=\"text\"  size=\"7\" style=\"text-align:center;border:groove ;  border-color:lightgray; padding: 2px;  border-width: thin; color:black; background:-webkit-gradient(linear, left top, right top, color-stop(0%,#BBBBBB), color-stop(50%,#F8F8F8), color-stop(100%,#BBBBBB)); width:107px;; height:38px;text-align: center;font-weight: bold;\" onChange='mango.view.setPoint("+ point.id +",\""+ pointComponent.id +"\",this.value);return false;' value="+ value +"%>";
return s;

---------------------------------------------------------------------------------------------------------------------------------------------
DISPLAY

/* Livre para uso sob a licença MIT  */

// Descrição do datapoint exibido. Oculta se omitida.
var descricao_do_datapoint = "";
// Definir prefixo e sufixo do valor (se necessário)
var prefixo = "";
var sufixo = "%";
var casas_decimais = 0;
// Personalizar estilo
var largura_minima = 107;
var altura_minima = 38;
var cor_fundo = "#e9e9e9";
var cor_descricao = "black";
var cor_valores = "black";
var tamanho_fonte = 15;
var usar_negrito = true;

// NÃO ALTERE A PARTIR DAQUI

// Definindo classe única para os elementos
var classeDp = "datapoint" + pointComponent.id;
// Formatando o valor com prefixo/sufixo
var valorFormatado = prefixo + value.toFixed(casas_decimais) + sufixo;
//var fonte = "Arial, sans serif";

// Criando elementos HTML
var s = "";
s += "<table class='" + classeDp + "'>";
// Primeira linha (descrição)
s += "    <tr> <td>" + descricao_do_datapoint + "</td> </tr>";
// Segunda linha (valor)
s += "    <tr> <td>" + valorFormatado + "</td> </tr>";
s += "</table>";

// Criando estilos CSS
s += "<style>";

// Definindo fundo do datapoint
s += "." + classeDp + " {";
s += "    height: " + altura_minima + "px;";
s += "    width: " + largura_minima + "px;";
s += "    background:-webkit-gradient(linear, left top, right top, color-stop(0%,#BBBBBB), color-stop(50%,#F8F8F8), color-stop(100%,#BBBBBB));";
s += "    border-collapse: collapse;";
s += "    border: 1px solid;";
s += "    border-color: gray;";
s += "}";

// Configurando fonte
s += "." + classeDp + " tr {"
s += "    text-align: center;";
s += "    font-size: " + tamanho_fonte + "px;";
//s += "    font-family: " + fonte + ";";
if (usar_negrito == true) {
	s += "font-weight: bold;";
}
s += "}";

// Configurando padding dos textos
s += "." + classeDp + " tr:nth-child(1) td {";
if ((descricao_do_datapoint.length > 0) && (typeof descricao_do_datapoint != "undefined")) {
	s += "    padding: 2px;";
	s += "    padding-bottom: 0px;";
} else {
	s += "    padding: 0px;";
}
s += "}";
s += "." + classeDp + " tr:nth-child(2) td {";
s += "    padding: 1px;";
s += "}";

// Definindo a cor dos textos
s += "." + classeDp + " tr:nth-child(1) {";
	s += "color: " + cor_descricao + ";";
s += "}";
s += "." + classeDp + " tr:nth-child(2) {";
s += "    color: " + cor_valores + ";";
s += "}";

s += "</style>";
return s;

---------------------------------------------------------------------------------------------------------------------------------------

LEVEL BAR

/* Livre para uso sob a licença MIT  */

// Defina o valor máximo e mínimo da escala
var maximo = 100;
var minimo = 0;
// Personalizar a barra
var altura = 120;
var largura = 60;
var cor_barra = "#0000A0";
// Modo "vertical" ou "horizontal"
var modo = "vertical";
var inverter_sentido = false;

// NÃO ALTERE A PARTIR DAQUI

// Criação de uma classe única para cada elemento
var classePai = "barraPai" + pointComponent.id;
var classeFilho = "barraFilho" + pointComponent.id;

// Conversão da escala do ponto para porcentagem
var porcentagem = 0;
if (value > maximo) {
	porcentagem = 100;
} else if (value < minimo) {
	porcentagem = 0;
} else {
	porcentagem = ((value - minimo) / (maximo - minimo)) * 100;
}
var porcentagem_inv = 100 - porcentagem;

// Criação dos elementos HTML
var s =  "<div class='" + classePai + "'><div class='" + classeFilho + "'></div></div>";
    s += "<style>";
    s += " ." + classePai + " {";
    s += "background: -webkit-gradient(linear, left top, right top, color-stop(0%,#BBBBBB), color-stop(50%,#F8F8F8), color-stop(100%,#BBBBBB));";
    s += "border:			solid 1px;";
    s += "border-color: 	black;";
    s += "height: " + altura + "px;";
    s += "width:	" + largura + "px;";
    s += "}";
    
    s += " ." + classeFilho + " {";
    s += "background: -webkit-gradient(linear, left top, left bottom, color-stop(0%,#05F2F2), color-stop(100%,#0477BF));";
    s += "position:		relative;";
	if (modo == "horizontal") {
			s += "height:		100%;";
			s += "width: " + porcentagem + "%;";
	} else {
			s += "height: " + porcentagem + "%;";
			s += "width:		100%;";
	}
	if ((inverter_sentido == true) && (modo == "horizontal")) {
		s += "left: " + porcentagem_inv + "%;";
	} else if ((inverter_sentido == false) && (modo == "vertical")) {
		s += "top: " + porcentagem_inv + "%;";
	}
	s += "</style>";
return s;

Limraj, 2023-07-12:
You already know that the cause is in the code, use the 'Inspect' function in chrome on the object to see which css is causing it.

GabrielTec251, 2023-07-12:
I am not able to identify, do I want gson to test the application?

Limraj, 2023-07-12:
1. Try ctrl+shift+r on page (reset browser cache)
2. The 'position' parameter is suspicious, and the places where you set height/width - I'd start by rigidly setting the width/height, next change position attribute;

Limraj, 2023-07-13:
Hi @GabrielTec251, have any other question?

closed 2023-07-17 by Limraj