SCADA-LTS/Scada-LTS #2615: BUG IN HTML COMPONENT TO INDICATE DATE AND TIME

GabrielTec251, 2023-07-12:
**
Using the html component, apply a code where as a result I have the date and time, but after a period the component starts to present a bug.

https://github.com/SCADA-LTS/Scada-LTS/assets/138117880/d230657b-08ce-4824-af7d-88efe63bd977

In the upper left corner of the graphic area that appears in the video.

COMPONENT CODE

<html>
	<body>
		<div id="data"style="font-family: 'arial', 'avant garde'; font-size: 15px;align:'center'; padding: 2px; width:150px;"></div>
		
		<div id="hora"style="font-family: 'arial', 'avant garde'; font-size: 15px;align:'center';width:150px;"></div>
	<body>
		<script>
			// Função para formatar 1 em 01
			const zeroFill = n => {
				return ('0' + n).slice(-2);
			}

			// Cria intervalo
			const interval = setInterval(() => {
				// Pega o horário atual
				const now = new Date();

				// Formata a data conforme dd/mm/aaaa hh:ii:ss
				const data = ' Data: ' + zeroFill(now.getUTCDate()) + '/' + zeroFill((now.getMonth() + 1)) + '/' + now.getFullYear();
        const hora = 'Hora: ' + zeroFill(now.getHours()) + ':' + zeroFill(now.getMinutes()) + ':' + zeroFill(now.getSeconds());

				// Exibe na tela usando a div#data-hora
				document.getElementById('data').innerHTML = data;
				document.getElementById('hora').innerHTML = hora;
			}, 1000);
		</script>
	</body>
</html>

desktop: dell optiplex 3000
OS: win 11
browse: chrome.

text translated by google translator.

Limraj, 2023-07-12:
Hi @GabrielTec251,
It's not an application bug, try doing it like this:
https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Date/toLocaleDateString

Regards,
Kamil Jarmusik

GabrielTec251, 2023-07-12:
opening the same application using the EDGE browser, it did not show this error,

On the left the application is on edge and on the right on chrome
[image]

Limraj, 2023-07-12:
We are unable to do anything about it, as you can see, the operation depends on the implementation of the browser, so you need to do it in such a way that it works on the browsers you care about. We use firefox/chrome. If you do it as described above, it will probably work fine on both browsers.

Limraj, 2023-07-12:
Try: https://stackoverflow.com/questions/5465375/javascript-date-regex-dd-mm-yyyy

Limraj, 2023-07-12:
```
<script>
	const interval = setInterval(() => {
		// Pega o horário atual
		const now = new Date();

                // Formata a data conforme dd/mm/aaaa hh:ii:ss
               const data = ' Data: ' + now.toLocaleDateString("en-GB");
               const hora = 'Hora: ' + now.toISOString().split("T")[1].split(".")[0];

                // Exibe na tela usando a div#data-hora
                document.getElementById('data').innerHTML = data;
                document.getElementById('hora').innerHTML = hora;
       }, 1000);
</script>
```

Limraj, 2023-07-13:
Hi @GabrielTec251 We're closing?

For this type of question it's worth using: https://github.com/SCADA-LTS/Scada-LTS/discussions

closed 2023-07-17 by Limraj