SCADA-LTS/Scada-LTS #2629: access to alarm properties

GabrielTec251, 2023-07-20:
I would like a way to get alarm properties, such as
*ALARM ACKNOWLEDGED
*ACTIVE ALARM
* INACTIVE ALARM

Reason, I would like to use it in graphic logics where for example:

In case of failure, the component must flash yellow,
 when the alarm is acknowledged but still in fault, the component is in solid yellow,
when the fault is over, the component switches to normal operating status.

In the scada: elipse e3 software, they have a browser where we can make requests for properties to use in scripts.
[image]

I imagine that it is possible in scada to do this search for properties to be able to apply it in scripts.

For example:

Var = (alarm name).AlarmStates ;

Something in that feeling.
[image]

Limraj, 2023-07-20:
@fabiodurao @GabrielTec251,
I don't remember that we had such functionality, maybe it could be handled with some configuration, but at the moment nothing specific comes to mind, if you Fabio have an idea, it's worth sharing. Anyway, it's an interesting direction of change, I think it's worth opening an issue.

GabrielTec251, 2023-07-25:
there is a tutorial on the scadaBr forum on how to access datapoint java classes in an easy way.

function dataPointInfo(identifier) {
	var dpDAO = new com.serotonin.mango.db.dao.DataPointDao();
	var pvDAO = new com.serotonin.mango.db.dao.PointValueDao();
	var dpVO = dpDAO.getDataPoint(identifier);
	var sdf = new java.text.SimpleDateFormat("yyyy-MM-dd HH:mm:ss");
	
	var tipos = {
		0: "Desconhecido",
		1: "Binário",
		2: "Multiestados",
		3: "Numérico",
		4: "Alfanumérico",
		5: "Imagem"
	};
	
	// Propriedades do data point
	this.id = parseInt(dpVO.getId());
	this.xid = String(dpVO.getXid());
	this.nome = String(dpVO.getName());
	this.codigoTipo = parseInt(dpVO.getPointLocator().getDataTypeId());
	this.tipo = tipos[this.codigoTipo];
	this.setavel = String(dpVO.getPointLocator().isSettable()) == "true" ? true : false;
	
	// Converte objetos Java para Javascript
	this.formataValor = function(valor) {
		switch (this.codigoTipo) {
			case 1:
				return String(valor) == "true" ? true : false;
			break;
			
			case 3:
				return Number(valor);
			break;
			
			case 2:
			case 4:
				return String(valor);
			break;
			
			case 0:
			case 5:
				throw new "Erro: Tipo inválido";
			break;
		}
	};
	
	// Último valor do data point
	this.valor = function() {
		return this.formataValor(pvDAO.getLatestPointValue(this.id).value);
	};
	this.ultimoValor = this.valor;
	
	// Último tempo do data point (humanamente legível)
	this.tempo = function() {
		return String(sdf.format(pvDAO.getLatestPointValue(this.id).time));
	};
	this.ultimoTempo = this.tempo;
	
	// Último tempo do data point (timestamp)
	this.tempoBruto = function() {
		return parseInt(pvDAO.getLatestPointValue(this.id).time)
	};
	this.ultimoTempoBruto = this.tempoBruto;
	
	// Últimos X valores do data point
	this.ultimosValores = function(maxValues) {
		var valoresJava = pvDAO.getLatestPointValues(this.id, maxValues);
		var valoresJS = new Array();
		for (var i = 0; i < valoresJava.size(); i++) {
			valoresJS.push(this.formataValor(valoresJava.get(i).value));
		}
		
		return valoresJS;
	};
	
	// Tempo das últimas X mudanças de valor do data point (humanamente legível)
	this.ultimosTempos = function(maxValues) {
		var temposJava = pvDAO.getLatestPointValues(this.id, maxValues);
		var temposJS = new Array();
		for (var i = 0; i < temposJava.size(); i++) {
			temposJS.push(String(sdf.format(temposJava.get(i).time)));
		}
		
		return temposJS;
	};
	
	// Tempo das últimas X mudanças de valor do data point (timestamp)
	this.ultimosTemposBrutos = function(maxValues) {
		var temposJava = pvDAO.getLatestPointValues(this.id, maxValues);
		var temposJS = new Array();
		for (var i = 0; i < temposJava.size(); i++) {
			temposJS.push(parseInt(temposJava.get(i).time));
		}
		
		return temposJS;
	};
	
	this.valoresEntre = function(startTime, endTime) {
		var valoresJava = pvDAO.getPointValuesBetween(this.id, startTime, endTime);
		var valoresJS = new Array();
		for (var i = 0; i < valoresJava.size(); i++) {
			valoresJS.push(this.formataValor(valoresJava.get(i).value));
		}
		
		return valoresJS.reverse();
	};
	
	this.temposEntre = function(startTime, endTime) {
		var temposJava = pvDAO.getPointValuesBetween(this.id, startTime, endTime);
		var temposJS = new Array();
		for (var i = 0; i < temposJava.size(); i++) {
			temposJS.push(String(sdf.format(temposJava.get(i).time)));
		}
		
		return temposJS.reverse();
	};
	
	this.temposBrutosEntre = function(startTime, endTime) {
		var temposJava = pvDAO.getPointValuesBetween(this.id, startTime, endTime);
		var temposJS = new Array();
		for (var i = 0; i < temposJava.size(); i++) {
			temposJS.push(parseInt(temposJava.get(i).time));
		}
		
		return temposJS.reverse();
	};
	
	this.valorEm = function(time) {
		var pointValue = pvDAO.getPointValueAt(this.id, time);
		if (!pointValue)
			pointValue = pvDAO.getPointValueBefore(this.id, time);
		
		return this.formataValor(pointValue.value);
	};
}

------------------------------------------------------------

link: http://forum.scadabr.com.br/t/simplificando-a-leitura-de-valores-via-script-no-scadabr/4085

----------------------------------------------------------

this way it is possible to get different information from the data point, using this script at the end of the test,

I was wondering if it was possible to develop something like this.