SCADA-LTS/Scada-LTS #2653: in not making sounds:

GabrielTec251, 2023-08-08:
in not making sounds:

it emits an alarm sound but the sound is not repeated, in this case it does not have a constant beep.

https://github.com/SCADA-LTS/Scada-LTS/assets/138117880/f3b0df64-3c16-48d6-bab0-312e205ab841

GabrielTec251, 2023-08-08:
right, in this case, would there be somewhere inside the application where I could configure this? in case any file folder, because here in my region this type of alarm is common, would you indicate where I could insert a loop?

thank you for your attention

Limraj, 2023-08-08:
I think that an option could be added in the system settings regarding whether the alarm is to be signaled continuously or once, or to do it per alarm level, e.g. alarms below a certain level are not important enough to be signaled continuously. ..

GabrielTec251, 2023-08-08:
it would be perfect, from urgent onwards it could be the continuous bep until the alarm is inactive or acknowledged.

Limraj, 2023-08-08:
In Scada-LTS, you can also set mute for a given event (Such an event is ignored in determining the highest alarm level), but what we are talking about will allow you to configure it globally, single, continuous, including total disabling, and I think we could add it per alarm level. However, I do not promise that we will deal with it in the near future ... Another thing is that if you listened to my request and tested this function before closing the topic, it might already work like this.

GabrielTec251, 2023-08-08:
yes yes, it was the first thing I tested when I installed the new version.

it only gives a bep for about 3s and stops, but as you said it will be perfect.

by the way, this translation is done by google translator, maybe it is subject to some misinterpretation by both parties,

in any case, thank you for the attention you are giving to the community.

Limraj, 2023-08-08:
I'm talking about the fact that under the topic you reported and related to this sound, I asked for a test before closing the topic, then you can still change/improve something. Therefore, if I ask for such tests, it is worth not ignoring it.

GabrielTec251, 2023-08-08:
resolved, with the help of celso, a collaborator of scadaBR, presented the following modifications:

part 01

in the file "pag.tag"

on line 174, add it inside if( alarmLevel >0) { mango.soundPlayer.play("level"+ alarmLevel);

it will look like this:

if (alarmLevel > 0) {
mango.soundPlayer.play("level"+ alarmLevel);
                        maxAlarmLevel = alarmLevel;
                        document.getElementById("__header__alarmLevelText").innerHTML = response.alarmlevel;
                        setAlarmLevelImg(alarmLevel, "__header__alarmLevelImg");
                        setAlarmLevelText(alarmLevel, "__header__alarmLevelText");
                        document.getElementById("__header__alarmLevelDiv").style.visibility='visible';
                        document.getElementById("__header__alarmLevelImg").style.visibility='visible';
                    }

then in the else you add the following:

mango.soundPlayer.stop();

in the end it will look like this:

 if (alarmLevel > 0) {
mango.soundPlayer.play("level"+ alarmLevel);
                        maxAlarmLevel = alarmLevel;
                        document.getElementById("__header__alarmLevelText").innerHTML = response.alarmlevel;
                        setAlarmLevelImg(alarmLevel, "__header__alarmLevelImg");
                        setAlarmLevelText(alarmLevel, "__header__alarmLevelText");
                        document.getElementById("__header__alarmLevelDiv").style.visibility='visible';
                        document.getElementById("__header__alarmLevelImg").style.visibility='visible';
                    }
                    else {
                        document.getElementById("__header__alarmLevelText").innerHTML = "";
                        document.getElementById("__header__alarmLevelImg").style.visibility='hidden';
                        document.getElementById("__header__alarmLevelDiv").style.visibility='hidden';
                        mango.soundPlayer.stop();
                    }

----------------------------

part 2

still in "pag.tag"

line 199, line was deleted

mango.soundPlayer.playOnce("level"+ alarmLevel);

-----------------------------

and the result is this: the alarm is continued until it is acknowledged or silenced.

https://github.com/SCADA-LTS/Scada-LTS/assets/138117880/9fa1f9ba-313d-46f5-b540-9fd55c87aa26

closed 2023-08-08 by Limraj