SCADA-LTS/Scada-LTS #2684: error in the report when writing message in the alarm

GabrielTec251, 2023-09-03:
when an event is triggered, in the events screen there is the option to add a message to the alarm and this message appears in the report, in scada br this function works, but in lts when you write a message in the alarm and generate a report, it ends giving an error in the system, I already did the tests and that's really what triggered the bug.

see the steps:

1. trigger an event on a datapoint;

2. go to the alarms screen and add a message to this event.

3. the data point that contains the event to a report.

4. open the report;

5. it is worth noting that when you do this, scada lts starts to show errors
[image]
[image]
[image]
[image]
[image]

browser: google chrome

lts version: 2.7.5.4

op: windows 11

Limraj, 2023-09-04:
Hi @GabrielTec251,
thank you for this report, it will be fixed in 2.7.6.1.

Regards,
Kamil Jarmusik

Limraj, 2023-09-04:
@GabrielTec251 
If you want to help, you can test this function:

1. Install Docker Compose:
[Docker Compose Windows installation](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Windows#4-docker-compose-)
[Docker Compose Linux installation](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Linux#4-docker-compose-)

2. How use commands:
[Docker Compose Tutorial](https://github.com/SCADA-LTS/Scada-LTS/wiki/Scada-LTS-docker-compose-tutorial#Commands)

3. Download file [docker-compose.yml](https://github.com/SCADA-LTS/Scada-LTS/blob/develop/docker-compose.yml) and change property image on **pr-2683** (from latest):
`image: scadalts/scadalts:pr-2683`
**pr-2683** is the pullrequest identifier:
https://github.com/SCADA-LTS/Scada-LTS/pull/2683

Regards,
Kamil Jarmusik

closed 2023-09-05 by Limraj