SCADA-LTS/Scada-LTS #2709: erro de http

GabrielTec251, 2023-09-21:
**Describe the bug**
A clear and concise description of what the bug is.
[image]
this is showing straight away

Limraj, 2023-09-22:
Hi @GabrielTec251, 
This message appears if the user is, for example, logged out of the application. The session lasts 30 minutes. If you refresh the page, do you not have to log in?

Regards,
Kamil Jarmusik

Limraj, 2023-09-26:
@GabrielTec251 
Okay, no answer, so I assume the matter is resolved. I'm closing the topic.

closed 2023-09-26 by Limraj

GabrielTec251, 2023-09-26:
Hi, I'm sorry for the delay, but this error happens due to having inframes inside the scada lts itself, in other words, I opened a scada lts page inside another page using it as a pop up, inside an inframe