SCADA-LTS/Scada-LTS #2721: Alarm pop-up while remote access using VPN

eng-rodrigo-lobo, 2023-10-06:
**Describe the bug**
When I try to access the ScadaLTS server via VPN, after a few seconds, the system throws a pop-up with the following warning: "Connect error:Whoops! Lost connection to http://scadaLTS_VPN_ip:8080/Scada-LTS/ws-scada/alarmLevel". 

**To Reproduce**
Steps to reproduce the behavior:
1. Login
2. Wait a few seconds

**Expected behavior**
I believe it shouldn't throw any alarms. After clicking OK, the system still works, doesn't lose connection, the gifs and scripts keep rolling, but for the final user it's a bummer having to click every time they open a new window.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Edge and Chrome
 - Version: ScadaLTS 2.7.5.4

Limraj, 2023-10-09:
Hi @eng-rodrigo-lobo,

1. After logging in as admin, right-click on 'Inspect' and show what is happening in the 'Network' tab - whether these connections are rejected;
2. After logging in as admin, right-click on 'Inspect' and show what is happening in the 'Console' tab;
3. Do you have a graphical interface and a Firefox/Chrome browser installed on the server where Scada-LTS is located and can you check whether Scada-LTS works properly on this server?

Regards,
Kamil Jarmusik

weschenfelder, 2023-10-09:
Hello, I hope you all are well!

I am facing problems when using DNS for external access to the system, I noticed that WebSocket only works when accessed locally via IP as per the examples below.

[image]

[image]

I believe it's just some route, because if I switch to the old interface the system works correctly without WebSocket errors.

Thank you in advance for your attention

eng-rodrigo-lobo, 2023-10-09:
> Hi @eng-rodrigo-lobo,
> 
> 1. After logging in as admin, right-click on 'Inspect' and show what is happening in the 'Network' tab - whether these connections are rejected;
> 2. After logging in as admin, right-click on 'Inspect' and show what is happening in the 'Console' tab;
> 3. Do you have a graphical interface and a Firefox/Chrome browser installed on the server where Scada-LTS is located and can you check whether Scada-LTS works properly on this server directly?
> 4. It is also worth installing the latest available prerelease;
> 
> Regards, Kamil Jarmusik

Hey @Limraj , first of all thanks for your reply!

Here are the answers to your questions:

1) Here is the "network" tab of my browser (Microsoft Edge) right after the error:
[image]<br>

2) Here is the "Console" tab right after the error:
[image]

I realized there's a situation with the Websocket connection, like @weschenfelder presented too.

3) Yes, I'm accessing the computer remotly via Anydesk. I set up all the system via Anydesk and it's running smooth locally, no exception thrown like when I try the remote access via VPN.

4) I'm considering this option, but I'd like to confirm if it's something we can solve or at least bypass for the final user first.

Thanks again for your time!

Limraj, 2023-10-10:
Hi @eng-rodrigo-lobo and @weschenfelder,

We use Scada-LTS via VPN, we also used after DNS, and we did not observe such problems, so probably the problem is related to network configuration... 

Why do ws-scada links show iframe.html at the end, this is not our link.... 

Regards,
Kamil Jarmusik

weschenfelder, 2023-10-10:
I believe that the problem occurs due to the reverse proxy, where it collects the DNS entry and points to a local IP and port. When routing directly via public IP, access to the system works perfectly.

However, because we host several services, websites and other applications, we use the incoming proxy and alongside it the WAF to fragment brute force attacks on the main host.

I will try to identify a way to resolve the reverse proxy issue and comment on the ticket again.

eng-rodrigo-lobo, 2023-10-13:
Hi @Limraj !

To be honest, I don't know very much about network connections, so I don't know exactly what should I check in the network configuration to check what may be causing this behaviour. If you have at least a few general directions to provide, I'd gladly accept.

About the iframe.html, I didn't even know it wasn't supposed to show there. The only add-on I configured in this system was a script for using Google Charts with "server scripts" in ScadaLTS, since the available graphs don't have other features that I usually use in my projects (change background colors, donut graphs, bar graphs, etc). 

To implement this, I followed a tutorial found at the ScadaBR Forum, adding a Script line at the page.tags file. Do you think could be related to the problem?

Limraj, 2023-10-16:
Hi @eng-rodrigo-lobo,
1. What happens when you go back to the old page.tag file? (console/network)
2. Send a link to this tutorial and write how you modified this page.tag file.
3. We certainly don't have any resources ending with iframe.html, perhaps this tutorial also includes information about placing the iframe.html file on the server.

Regards,
Kamil Jarmusik

eng-rodrigo-lobo, 2023-10-18:
Hi @Limraj 

Just to add on my last post, I was curious about what was calling this iframe thing. This time I updated the ScadaLTS to the latest version 2.7.6 and accessed via VPN this updated ScadaLTS, getting again the same results.

When I checked the iframe calls, I realized it comes from the following path within the ScadaLTS folders:
[image]

Do you know what triggers the ScadaLTS to call this resource?

Regards,
Rodrigo Lobo

Limraj, 2023-10-19:
Hi @eng-rodrigo-lobo 
Looks like a problem with the old version of the sockjs library and Internet Explorer:
https://github.com/sockjs/sockjs-client/issues/249
So there is a chance that you have an old version in the browser cache, since Scada-LTS 2.7.1.1 we use sockjs version 1.6.0.

To solve this:
1. After the page loads, try -> Shift + F5 or Ctrl + Shift + r ('Reload the current page, ignoring cached content')
2. If that doesn't help, maybe there is a compatibility issue with sockjs and edge, then just install chrome or firefox.

Regards,
Kamil Jarmusik

Limraj, 2023-10-27:
Hi @eng-rodrigo-lobo, 
What is the situation?

eng-rodrigo-lobo, 2023-10-30:
Hi @Limraj !

Busy days lately, sorry for the late answer.

I tried cleaning the cache and tried to access the ScadaLTS using Firefox and Chrome (both with the latest releases) and still got the same issue...

One thing I realized is that a websocket tries to connect and fails always,and then the pop-up is shown, I'll show you below:

MICROSOFT EDGE (version 118.0.2088.76)
[image]

GOOGLE CHROME (version 118.0.5993.118)
[image]

FIREFOX (version )
[image]

So, I still have no clue of what's going on

Limraj, 2023-10-30:
Hi @eng-rodrigo-lobo,
It's not about clearing the cache, but refreshing the page with that exact shortcut.

Check user sockjs version:
http://host:8080/Scada-LTS/resources/node_modules/sockjs-client/dist/sockjs.min.js
[image]

Regards,
Kamil Jarmusik

eng-rodrigo-lobo, 2023-10-30:
Hi @limraj 

I checked here, sockjs version is v1.6.0, as below (this was in Firefox):
[image]

To keep a pattern, I tried both commands you suggested (shift + F5 and ctrl +shift + R) at the watchlist page, but the result was the same (problem with connecting to the websocket and alarm pop-up).

weschenfelder, 2023-10-30:
Hello @eng-rodrigo-lobo, this same problem occurred when I imported a JSON completely from another installation, I went in search of delving deeper into the solution to this error and the only way to solve it was to delete the existing version of both java and tomcat and reinstall them.

According to the java forums, this bug can occur due to several problems, among them there may have been a failure during the installation that only became apparent now that you are using the method mentioned above.

eng-rodrigo-lobo, 2023-11-06:
Hi @Limraj 

I was using a partner's VPN to test the remote access, as I'm developing a SCADA for this company and they'll use this network for remote access of the system.

Today I decided to try another VPN just to be sure and for my surprise, it worked fine! No alarm pop-ups.

This new VPN was just a standart OpenVPN CloudConnexa service (https://openvpn.net/cloud-vpn/  | free-tier, up to 3 connections), I just created an account, installed the clients at the 2 testing PCs, configured the firewall exception rules for the ScadaLTS 8080 port and voilà.

I'll check with my partner's VPN provider what are the settings, if they provide me this information, and I'll try to compare with the standart setting of OpenVPN to see if we realize what could be causing those errors mentioned before.

Anyways, I guess your initial assumption was right, probably was some network configuration.

closed 2024-07-08 by Limraj