SCADA-LTS/Scada-LTS #2722: MQTT data source stops communicating with broker after PC restarts or comes back from sleep mode

eng-rodrigo-lobo, 2023-10-10:
**Describe the bug**
The MQTT datasource stops communicating with the configured broker after the PC restarts or comes back from sleep mode. It only returns to normal if we switch it 'off' and back 'on' again manually.

**To Reproduce**
1. Set up a MQTT data source
2. Let it run for a few minutes to check if it's working fine
3. Restart the computer or put it on sleep mode
4. Check the last timestamp of the data points of the MQTT data source and you'll see it holds the last info before the restart/sleep mode

**Expected behavior**
Well, as all the other data sources, it should restablish communication automatically as the server is on again.

**Desktop:**
 - OS: Windows 11
 - Browser: Microsoft Edge, Chrome and Firefox
 - Version: Scada-LTS 2.7.5.2 and Scada-LTS 2.7.5.3 (I haven't checked for the latest releases)

Thanks in advance for your support and attention.

Limraj, 2023-10-10:
Hi @eng-rodrigo-lobo,
You mean you have a broker on one computer, that computer goes into sleep mode, or you reboot it, and the MQTT data source in Scada-LTS on another computer does not bear the connection to that broker, when does it go back to normal?

Regards,
Kamil Jarmusik

eng-rodrigo-lobo, 2023-10-13:
Hi @Limraj !

Sorry, I think I didn't give a better overview of the application.

I have a smartmeter at home that provides the information via MQTT. I configured the smartmeter to publish the data to emqx.io (a broker service online). I have configured the Scada LTS in my test bench to subscribe to this topics, so the broker is virtually never down. The problems that I presented before were only at the SCADA LTS side, after the Scada PC went on sleep mode or restarted.

About the lastest releases, I understand and apologize for using old releases, by the time I was installing the system these were the only ones with the windows installer available. Is there an easy way to upgrade versions withou using the installer?

Limraj, 2023-10-16:
Hi @eng-rodrigo-lobo,
1. "It only returns to normal if we switch it 'off' and back 'on' again manually."
you wrote that if you turn data source off and on it works? An analogy to this is restarting an application(and system). During stop and startup, it does the same thing as during the operation of stoping and starting the Data Source. 

This seems to be a specific behavior of windows in sleep/hibernation state... 

2. Generally, in our Data Source MQTT you can configure up to 10 connection attempts after connection failure. In the next release we will allow an unlimited number (Integer.MAX = 2147483647) of times to configure these connection attempts when the connection is lost. 

3. When the Data Source is not working, do you have any errors, events related to this Data Source when you go to edit it? If so, can you show the content of these messages?

4, You can try with a different version of the protocol, they are supported by different libraries.
[image]

5. Upgrade Scada-LTS
You can replace the war file (in Scada-LTS/tomcat/webapps), but it is worth backing up the database first, returning from a newer to the previous version may be difficult because some versions involve changes in the database structure. 

Regards,
Kamil Jarmusik

eng-rodrigo-lobo, 2023-10-20:
Hi @Limraj !

Just to give you some feedback, I updated ScataLTS to the latest realease 2.7.6 and it is working fine.

Maybe it was some bug from older releases, but as I updated the software the datasource started working fine.

Thanks for your attention in this topic!

closed 2024-03-25 by Limraj