SCADA-LTS/Scada-LTS #2724: DataSource Create Button/Option not show

odenisnobre, 2023-10-15:
After creating some Datasources manually and via API I noticed that the button to create new datasources via graphical interface was no longer available.

[image]

odenisnobre, 2023-10-16:
[image]

[image]

The problem occur in other browsers.

[image]
[image]
[image]

Limraj, 2023-10-16:
1. Can you show what you have in: right click on the website -> inspect -> console tab?
2. What is node-env? Have you looked on the Internet what causes the error NS_ERROR_NET_PARTIAL_TRANSFER / NS_BINDING_ABORDED?
3. When did you notice this problem? I understand that it worked before, does it depend on the version?
4. If you restart tomcat on the server, does the problem still occur?
5. You can export the project using the Import/Export function then check if the problem is related to one of these Data Source types, remove all of them and add one data source type first, if ok then remove and add the second data source type, if ok then add another data source yes to be the situation you present with two data sources and check if the problem still occurs. If not, import the project with a previously exported file and check if the problem still occurs.

I recommend making annotations to the questions (Re./Ref. 1, Ref. 2, etc.) so that we can understand each other more easily.

Regards,
Kamil Jarmusik

odenisnobre, 2023-10-16:
Answers:
1: 
[image]

2: I didn't find anything related, at least I couldn't understand the context

3: In fact, the problem occurred right after creating a new datasource via API

4: Even restarting the container the error persists

5: The error occurs when adding an OPC DA type datasource

odenisnobre, 2023-10-16:
https://github.com/SCADA-LTS/Scada-LTS/assets/8119694/26a61f08-f434-4a0e-b931-798d6a98e93f

odenisnobre, 2023-10-16:
[image]

the error continues after updating to the new build

Limraj, 2023-10-16:
You may have problems configuring these connections in Docker, so I prepared a war for you, it is not a production version, file [Scada-LTS_pre_2.7.6.1_NO_PRODUCTION.zip](https://github.com/SCADA-LTS/Scada-LTS/releases/download/v2.7.6/Scada-LTS_pre_2.7.6.1_NO_PRODUCTION.zip):

First stop tomcat, and delete or move the old ScadaBR/Scada-LTS folders and war file from tomcat/webapps, then extract the downloaded file, it will be the war file, and move it to tomcat/webapps, and start tomcat.

odenisnobre, 2023-10-16:
I made some changes to my docker compose to make it easier to use the .war file, but now it doesn't load the application.
Where could the error be?

```
# Docker compose file to manage your deployed images.
# Use MySQL server 5.7 and latest Scada-LTS local build.
# Using attached webapps folder as developer you will be able to modify the static content from host os.
# Attach shell to stop the tomcat instance and then you will be able to run in JPDA mode.
version: '3'
services: 
    database:
        container_name: db
        image: mysql/mysql-server:8.0.32
        ports:
            - "3306:3306"
        environment: 
            - MYSQL_ROOT_PASSWORD=root
            - MYSQL_USER=root
            - MYSQL_PASSWORD=root
            - MYSQL_DATABASE=scadalts
        expose: ["3306"]
        volumes:
            - /home/noobres/volumes-docker/scadalts/mysqldata:/var/lib/mysql:rw
            - /home/noobres/volumes-docker/scadalts/mysqlconfig:/etc/mysql:ro
        command: --log_bin_trust_function_creators=1
    scadalts:
        container_name: scada
        image: scadalts/scadalts:pr-2683
        environment:
            - CATALINA_OPTS=-Xmx512m -Xms512m
        ports: 
            - "8080:8080"
        depends_on: 
            - database
        expose: ["8080", "8000"]
        volumes:
            - /home/noobres/volumes-docker/scadalts/aplicacao/logs/:/usr/local/tomcat/logs:rw
            - /home/noobres/volumes-docker/scadalts/aplicacao/app/:/usr/local/tomcat/webapps:rw
        links:
            - database:database
        command:
            - /usr/bin/wait-for-it
            - --host=database
            - --port=3306
            - --timeout=30
            - --strict
            - --
            - /usr/local/tomcat/bin/catalina.sh
            - run

odenisnobre, 2023-10-16:
I configured the new .war file but the error still persists.

odenisnobre, 2023-10-16:
[image]

Limraj, 2023-10-16:
@odenisnobre 
Have you done anything with this before (OPC DA)? Since when have you had a problem with this?

odenisnobre, 2023-10-16:
Well, let's go.

In short, the error occurs when creating the OPC-DA interface via UI or via API, the problem occurs both.
But in the end the error is not when creating the Datasource but when creating the DataSource tags

Limraj, 2023-10-16:
@odenisnobre 
1. If the problem occurs after creating this Data Source via UI, you need to answer my questions. If it breaks like this after adding this data source via UI, it would be very strange. 
"creating Data Source via UI" -> Open Scada-LTS in browser, login as admin, go to Data Sources, select "OPC DA" and click on 'Add':
[image]
Next set fields and click on 'Save';

3. "creating the DataSource tags"... can you explain it?

odenisnobre, 2023-10-16:
If you notice in the video below, you will see that the issue is during the creation of the points and not necessarily in the datasource

https://github.com/SCADA-LTS/Scada-LTS/assets/8119694/cb04fdfa-c9cb-4d45-a8e8-0d4ca8dd4109

Limraj, 2023-10-16:
@odenisnobre 
Before you create an issue, try to describe the problem as precisely as possible. I need written information, I always need logs (tomcat/logs), I need to add a script, step by step, then you can also add this video. See, we lost hours to determine that the problem concerns adding tags in OPC DA, which should be immediately described in the issue content. If you used OPC DA, API, whatever and we already talked about it, you need to go back to this information and make some reference, e.g. it stopped working for me with the version, or I didn't use it before.

odenisnobre, 2023-10-16:
Well, sorry about that.

I think you already understand the problem even with this confusion I made.

With the version you sent me, in addition to the button not being visible, other datasources are also not visible.

If you still need more information, let me know.

Again, sorry.

Limraj, 2023-10-16:
@odenisnobre It's not about apologizing, it's about effective communication.
You need to complete the information in the issue, give an appropriate title, add a script, etc. so that the person who comes across it after some time does not have to browse through these messages, and you did not provide all the information I asked for, so it also needs to be included there.

Limraj, 2023-10-16:
If you don't want to be helpful, don't count on others to give you their time.

odenisnobre, 2023-10-16:
The problem is not the usefulness but the confusion that is creating the case.
The closure request is due to the confusion that has arisen around something, which has already been analyzed, so simple.

I sincerely apologize for that.

If you think it's better and so as not to extend this case any further, we can remove it and start another with less confusion.

Limraj, 2023-10-16:
@odenisnobre You have noticed a problem that exists, you just need to add this information in the issue content and that's it, instead of getting offended. If the information is completed, there is a greater chance that someone will take care of it.

In addition to what you have already written, you need:
1. Tomcat logs;
2. Information whether you have used this data source before, because from what I remember you have already used it, or at least tried to help someone else?

If you don't want to do it in this issue, we can close it, but you need to create a new one with the information added. The more precisely you describe and test the problem, the greater the chance that someone will take care of it. You always need:

1. Scenario;
2. Logs;
3. Version, it is worth adding here if you have used this function before;

You can use our templates (bug) to make it easier to fill in this data.

odenisnobre, 2023-10-16:
OK, let's continue here.

An OPC-DA configuration test was previously carried out, but I believe it was in previous versions that I don't remember, well before version 2.7.6 build 59745851
[logs.zip](https://github.com/SCADA-LTS/Scada-LTS/files/12919809/logs.zip)
95

Below is the tomcat log.

Limraj, 2023-10-16:
You need to complete the content of the issue, not in the comments, this will increase the chances that someone will take care of it.

You can further investigate whether you can add these tags individually, whether they need to be enabled in order to crash, whether they can be disabled and you can check some more scenarios, then it will make the work easier for the developer. If you can, check on v2.6.18.

I would check it myself, but I don't have a simulator at hand.

odenisnobre, 2023-10-16:
I will test the scenarios in version 2.6.18.

Limraj, 2023-10-16:
Can you edit this issue and add important information that we wrote about, and corrected title? I will be grateful...

odenisnobre, 2023-10-16:
I'll do the tests and make the changes.

Limraj, 2023-10-16:
Thank you very much. :)

Limraj, 2023-10-16:
Okay,
It looks like I found the reason why loading the data source list fails after adding a point to Data Source OPC DA. So you don't need to complete it like that anymore, and the reason is not related to the tags. 
This applies to the configurationDescription field in Point Locator, which cannot be null, and in OPCPointLocatorVO it was null by default. This also applies to classes such as: 
- ASCIIFilePointLocatorVO, 
- ASCIISerialPointLocatorVO, 
- Dnp3PointLocatorVO, 
- DrStorageHt5bPointLocatorVO, 
- NodaveS7PointLocatorVO,
oraz:
- OPCPointLocatorVO

Similarly to how you tested earlier on Docker, you can test this pullrequest and let us know if it works OK, then we will merge to version 2.7.6.1.
https://github.com/SCADA-LTS/Scada-LTS/pull/2725

odenisnobre, 2023-10-16:
[image]

I tested with pr-2725 and the error persists.
Not only does the add datasource button appear but also the other datasources
[image]

Would you be able to confirm if the build pro pr-2725 is the same 6496147414

Limraj, 2023-10-16:
You ran on an old image, there is not other option.
[image]
[image]
[image]

odenisnobre, 2023-10-16:
Very strange.
I'll see what I did wrong here.

Limraj, 2023-10-16:
Are you using this file? Removed this line:
`- /home/noobres/volumes-docker/scadalts/aplicacao/app/:/usr/local/tomcat/webapps:rw`
in:
```
# Docker compose file to manage your deployed images.
# Use MySQL server 5.7 and latest Scada-LTS local build.
# Using attached webapps folder as developer you will be able to modify the static content from host os.
# Attach shell to stop the tomcat instance and then you will be able to run in JPDA mode.
version: '3'
services: 
    database:
        container_name: db
        image: mysql/mysql-server:8.0.32
        ports:
            - "3306:3306"
        environment: 
            - MYSQL_ROOT_PASSWORD=root
            - MYSQL_USER=root
            - MYSQL_PASSWORD=root
            - MYSQL_DATABASE=scadalts
        expose: ["3306"]
        volumes:
            - /home/noobres/volumes-docker/scadalts/mysqldata:/var/lib/mysql:rw
            - /home/noobres/volumes-docker/scadalts/mysqlconfig:/etc/mysql:ro
        command: --log_bin_trust_function_creators=1
    scadalts:
        container_name: scada
        image: scadalts/scadalts:pr-2683
        environment:
            - CATALINA_OPTS=-Xmx512m -Xms512m
        ports: 
            - "8080:8080"
        depends_on: 
            - database
        expose: ["8080", "8000"]
        volumes:
            - /home/noobres/volumes-docker/scadalts/aplicacao/logs/:/usr/local/tomcat/logs:rw
            - /home/noobres/volumes-docker/scadalts/aplicacao/app/:/usr/local/tomcat/webapps:rw
        links:
            - database:database
        command:
            - /usr/bin/wait-for-it
            - --host=database
            - --port=3306
            - --timeout=30
            - --strict
            - --
            - /usr/local/tomcat/bin/catalina.sh
            - run
```
Why didn't you do it according to the instructions I wrote to you earlier:

> Perhaps a fix will solve this problem (if this problem with OPC DA): #2715 in version 2.7.6.1
> 
> You can check it as follows:
> 
> 1. Install Docker Compose:
>    [Docker Compose Windows installation](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Windows#4-docker-compose-)
>    [Docker Compose Linux installation](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Linux#4-docker-compose-)

> 2. How use commands:
>    [Docker Compose Tutorial](https://github.com/SCADA-LTS/Scada-LTS/wiki/Scada-LTS-docker-compose-tutorial#Commands)
> 3. Download file [docker-compose.yml](https://github.com/SCADA-LTS/Scada-LTS/blob/develop/docker-compose.yml) and change property image on **pr-2683** (from latest):
>    `image: scadalts/scadalts:pr-2683`
>    **pr-2683** is the pullrequest identifier:
>    [Release/2.7.6.1 #2683](https://github.com/SCADA-LTS/Scada-LTS/pull/2683)
> 
> Regards, Kamil Jarmusik

odenisnobre, 2023-10-16:
Sensational.
Now yes.
It worked smoothly.

Limraj, 2023-10-16:
Do you mean the issue with Docker works, and what about the errors from this issue? With displaying a list of data source types and all data sources in the list?

> Sensational. Now yes. It worked smoothly.
> 
> I would like to externalize the Scada-LTS part so that I can make customizations.
> 
> ```
> - /home/noobres/volumes-docker/scadalts/aplicacao/app/:/usr/local/tomcat/webapps:rw
> ```
> 
> Would I have to download another .war file with this version of pr-2725?
That parameter maps the tomcat/webapps location in the container to that of the host, so as a host you could throw war there and see the effects in the container, the container then does not see the tomcat/webapps location on itself, but uses the host's location.

odenisnobre, 2023-10-16:
Understood.
Well, overall, the solution was perfect.
The issue of volume mapping will be left for later.

Excellent job.
Thank you very much.

odenisnobre, 2023-10-16:
One more thing, which file or version of the .war file can I have this new configuration?

I returned the volume mapping, but the .war file I added is still with the old build.

Limraj, 2023-10-17:
If you add the line:
`- /home/noobres/volumes-docker/scadalts/aplicacao/app/:/usr/local/tomcat/webapps:rw`
then the container ignores its location `/usr/local/tomcat/webapps`, so it will not load the war related to the image version, but it will load what you put in its location: `/home/noobres/volumes-docker/scadalts/aplicacao/app/`, so if you had an old war there, I'll load the old war. Removing this line causes the container to use location `/usr/local/tomcat/webapps` and load the version of war it has in it and results from the image version and then we have a new version. In order for a new war to be loaded with this setting, you would need to upload the new war to `/home/noobres/volumes-docker/scadalts/aplicacao/app/`. Then it doesn't matter what version of the base image is, because you decide about the uploaded version by uploading the appropriate version to your location. From what I remember, you needed something like this to freely modify Scada-LTS and for the changes to be visible from the container.

odenisnobre, 2023-10-17:
What version of .war is inside the image?
This is what I need to add to my volume to have the information updated.
Having the same version of .war as in the image, I can add it to:
>/home/noobres/volumes-docker/scadalts/aplicacao/app/

Limraj, 2023-10-17:
try command:
`docker cp <containerId>:/file/path/within/container /host/path/target`

odenisnobre, 2023-10-17:
Perfect, just what I needed.
It worked the way I wanted.
Thank you very much

Limraj, 2023-10-17:
Okay, so I'm closing the topic.

closed 2023-10-17 by Limraj