SCADA-LTS/Scada-LTS #2727: Alignment error on graphic screen with html component

GabrielTec251, 2023-10-18:
When an HTML component is inserted, a bug may occur, where it puts in decimal places or X and Y axis values, and even after manually rounding these values, it continues to repeat the error, as shown in the video below:

Note: this error only occurs after a certain amount of time

https://github.com/SCADA-LTS/Scada-LTS/assets/138117880/9f655a22-867d-4b76-8259-0aaac9fb30d6

scada-lts v2.7.5.4
win:10

fabiodurao, 2023-10-30:
I also had this problem after I imported a project, I believe it has nothing to do with the import. What I noticed was that this happened when I was zoomed in on the browser, as you can see in the video. When I left the zoom at 100% (ctrl+0), the coordinates returned to normal.

Patrykb0802, 2025-01-17:
This issue was fixed in pull request #1770

Limraj, 2025-01-27:
Hi @GabrielTec251,

We are unable to replicate this on v2.7.5.4 for the HTML component.
We are unable to get the x,y parameters with floating point values:

[image]
[image]

So he's probably using a different version of the Scada-LTS. You need to check this, or install the latest stable version 2.7.7.3, or the development version 2.7.8.

Regards,
Kamil Jarmusik

closed 2025-02-05 by Limraj