SCADA-LTS/Scada-LTS #2747: Updated installer for Ubuntu Server

weschenfelder, 2023-10-25:
Hello, could you create a new installer with the new update for Linux Ubuntu?

I missed a recent update, there are a lot of bugs occurring and I believe that the new version has already fixed it.

I opened the call as a problem because I believe this solves many people's problems.

weschenfelder, 2023-10-26:
I believe that the installation flow has changed, after a few hours I noticed that to carry out the installation, the interpretation of the .sh files must be corrected as an example

sed -i 's/\r//' tomcat_config.sh
sed -i 's/\r//' mysql_start.sh
sed -i 's/\r//'java_install.sh

After doing this, the
java_install.sh

And then only after that the .sh execution commands should be executed

I believe this occurs in versions above 20.04 LTS of the Ubuntu server.

Limraj, 2023-10-26:
Hi @weschenfelder,

Thank you for pointing out that I only replace the war file in the installer when updating, but unfortunately I recently used Windows system to replace the file and pack it, which could have caused the problem.

Brief information in the context of the installer, we have these two scripts there:
tomcat_start.sh and mysql_start.sh to update the application to the latest version, when we already have the database from the previous version of the installer, we only call tomcat_start.sh and while the installer is running, we provide the connection data, host, port, user corresponding to the server we want to use , so you don't have to use the database server from the installer, you can use an external server.

Regards,
Kamil Jarmusik

Limraj, 2023-10-26:
@weschenfelder I corrected it, let me know if it's ok now:
https://github.com/SCADA-LTS/linux-installer/releases/tag/v1.2.1

weschenfelder, 2023-10-27:
Hi @Limraj, I tested it and apparently the installer is working correctly now, thank you.

Would it be possible to create documentation on how to use a database on another virtual machine?

Example:

TomCat64
VM: 192.168.101.10

Mysql
VM:
192.168.101.11

I believe this would make it easier to implement large-scale projects.

fabiodurao, 2023-10-27:
In this context, I believe that the best thing is to just run the tomcat script and enter the username and password of this other server, it will try to link with localhost, after the installation is complete, it will not work, but then you edit the context.xml file and change from localhost to the database server IP and run the tomcat script again, it should start the service.

[image]

fabiodurao, 2023-10-27:
@Limraj , how does Scada-LTS behave (if we use the database from a previous installation) and are there any changes to the new version's database? Does it create what is missing or will we have a problem?

Limraj, 2023-10-27:
@fabiodurao, 
We have a migration mechanism that introduces changes to the database from subsequent versions, so as long as we change the version to a higher one, there should be no problems, but the problem will be if we want to go back to the previous version, so it is worth doing exports json and dumping the database before upgrading the application.

"In this context, I believe that the best thing is to just run the tomcat script and enter the username and password of this other server, it will try to link with localhost, after the installation is complete, it will not work, but then you edit the context.xml file and change from localhost to the database server IP and run the tomcat script again, it should start the service."

@Patrykb0802 updates the linux installer with the ability to insert the database host into tomcat, then file modification will not be necessary.

fabiodurao, 2023-10-27:
I don't know if it has already been fixed, but there is a problem with the "&" character inside the DB password, it separates it into two strings and causes problems with the connection, this is a problem when we generate random passwords with password vaults like bitwarden which Add special characters to make the password stronger. Today I visually inspect it before copying the password, if it has "&", I remove it before copying.
It would also be good to have a way to make the installer automatically start Scada-LTs along with Linux, perhaps leaving it as a service or in another way.

Limraj, 2023-10-27:
@fabiodurao this is also being corrected...
We will suspend the service for now, we want it to be as dependent on the system configuration as little as possible.

weschenfelder, 2023-10-30:
Hi @Limraj, could you tell me if this Mysql host situation is already updated in the linux-installer_v1.2.1_ScadaLTS_v2.7.6.1_Setup.zip file?

Limraj, 2023-11-02:
Hi @weschenfelder, We're working on the installer, we'll probably have it finished next week.

weschenfelder, 2023-11-17:
Hello @Limraj  @Patrykb0802, I hope to see you and the whole team well!

Could you tell me if there is a plan to release the version with this change?

I ask for this because I have a giant project coming up and I would like to implement this method.

Limraj, 2023-11-17:
Hello @weschenfelder,
We have created a release: https://github.com/SCADA-LTS/linux-installer/releases/tag/v1.2.2

We invite you to cooperate, our business contact: frederic.dumas@softq.pl ;

weschenfelder, 2023-11-20:
Hello @Limraj , I hope I see you well!

I tested it over the weekend and the installer keeps looping when it asks for the DB host, analyzing tomcat_config.sh due to the condition defined in the while for DATABASE_HOSTNAME loop.
Attached is the corrected script for installation.

[tomcat_config.zip](https://github.com/SCADA-LTS/Scada-LTS/files/13414889/tomcat_config.zip)

After this correction I also noticed that the migrations are executed but end up corrupting the database at some point, I tried to redo it a few times and it always happens that some information is missing, today during the day I will try to run the scripts manually to see how far I can get .

[image]

weschenfelder, 2023-11-22:
Good morning!

I'm sorry that my previous comment was so simple and didn't describe all the facts in detail. I will do my best to make sure this doesn't happen again.

Remember that I do what I can to help the ScadaLTS community, especially in Brazil, where licensing Windows-type operating systems is extremely expensive and not very accessible for small projects and individuals who are taking their first steps in the search for knowledge and tend to to look for open source solutions, such as Ubuntu, to host the system.

Today, I'll go into more detail about what happened during testing of build 2.7.6.1 and post a pull request with a possible fix.

Thank you very much in advance.

Limraj, 2023-11-22:
Hi @weschenfelder,
Sorry for these glitches with the installer, for the hostname field, it will be corrected using a special regex, and other regexes will also be corrected.

If you are presenting a problem, it is better not to assume anything, but to provide information that may enable reproduction or understanding of what is happening (logs, screenshots). Otherwise we can't help much.

Regards,
Kamil Jarmusik

Limraj, 2023-11-22:
Hi @weschenfelder,
Here is the corrected installer: https://github.com/SCADA-LTS/linux-installer/releases/tag/v1.3.3

Thank you for your participation.
I will be grateful for confirmation whether the installer works as expected.

If there is anything wrong, please write, especially what you noticed on 2.7.6.1.

Regards,
Kamil Jarmusik

weschenfelder, 2023-11-24:
Hi @Limraj, greetings!

I tested the installer and now it works perfectly.
I noticed that the changes made to REGEX were 100% functional.

thank you very much!

[image]

closed 2023-11-24 by weschenfelder