SCADA-LTS/Scada-LTS #2757: Gradle 8.4 jdk11

skyworker0725, 2023-11-06:
Cause: class jdk.internal.loader.ClassLoaders$PlatformClassLoader cannot be cast to class java.net.URLClassLoader (jdk.internal.loader.ClassLoaders$PlatformClassLoader and java.net.URLClassLoader are in module java.base of loader 'bootstrap')

Patrykb0802, 2023-11-07:
Hi @skyworker0725,
Currently Scada-LTS is using Gradle version 7.1 and this is why this error may occur. You can look up how to change Gradle version in your project on our wiki for both [Windows](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Windows#3-intellij-) and [Linux](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Linux#3-intellij-). This information is mentioned in the last point in the IntelliJ paragraph.

Greetings,
@Patrykb0802

closed 2023-11-09 by Patrykb0802

RenatoExpert, 2023-12-02:
@Patrykb0802 may we upgrade gradle to 8?