SCADA-LTS/Scada-LTS #2763: Scripting : list of already selected context points does often not appear on the first click on the script

nduvernois, 2023-11-07:
From any tab, click on scripting, and then on an existing script. Quite often (probably on the first use after PC startup), the list of already selected datapoints does not appear. It is needed to open another script, and then come back to the actually targeted script to have the proper display of already selected datapoints

OS : Windows Server 2016
Browser : Chrome 118
ScadaLTS : 2.6.18

ga7x, 2024-02-02:
@nduvernois: Hi, following these tips I am not able to find the problem at the moment. Does the problem still occur?
I implemented the following scenarios:
- I created points,
- I created a list of data points,
- The created list of points is usually displayed correctly after the first launch of the Script tab.
- Additionally, I added a new point in Virtual Data Source and went to Scripts. I checked that the newly added point appears in the list of points.

nduvernois, 2024-02-02:
Hello, I just retried and it works correctly. So ... I don't know. I'm actually not using the official 2.6.18 release, but a special package prepared by @Limraj (currently build 3789793770). Maybe the problem has been corrected by some fixes he introduced. Maybe it is or was influenced by the size of the database (# 16 000 datapoints, # 50 scripts).

ga7x, 2024-02-12:
I am unable to recreate this scenario. Therefore, I am closing this issue.

ga7x, 2024-02-12:
Hello, I can't reproduce this scenario. Therefore, I am closing this issue.

closed 2024-02-12 by ga7x