SCADA-LTS/Scada-LTS #2765: Add scatter plot feature

nduvernois, 2023-11-08:
Typically for energy monitoring, it can be useful to plot Y versus X (e.g. heating consumption versus degree-days). Scada-LTS currently allows to plot on the same graph X and Y versus time.

Solution would be to create a new kind of graph : plot Y(t) versus X(t) during a selectable period of time, with 1 dot for each (x,y) combination.

Limraj, 2024-05-06:
Hi @nduvernois,
I understand. But isn't this time necessary in most cases? And doesn't the Consolidated chart feature with the superimposition of heating consumption and degree-days charts solve this problem, and we can see it over time?

Regards,
Kamil Jarmusik

nduvernois, 2024-05-13:
> - External - This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

INTERNAL & PARTNERS

Hello Kamil

I reply to your comment by e-mail because, unfortunately, Forvia firewall does not allow me to add pictures in Github.
So, what I can currently display in ScadaLTS is this (in this case with 1 point every day over 1 month, but I could obviously also plot 1 point every month over 1 year):
***@***.***
So, if the red curves does not stay between the blue & the green ones, something has happened.

As a comparison, here are the graphs that I use in an Excel file, with 1 point per month over many years :
***@***.***

The left graph more or less matches the ScadaLTS current capability (Y vs t). The right graph shows exactly the same data, but Y vs X. And in this case, I see clearly that year 2019 was on a globally different trend. In this case, I know why : we switched from neon lights to LED’s. (dots on the Y=0 axis simply mean that these are data from the future 😊).
Let’s now see this other graph :
***@***.***

Here, since the beginning of the year, I am clearly very significantly out of the expected bounds (left graph), but the right graph also tells me that the relationship between Y and X is probably very different, which is another significant trouble.

So, I am convinced that adding this scatter plot feature would help analysis of data.

Kind regards

Nicolas Duvernois
CLEAN MOBILITY
Bavans Site + R&D Division + FHS Division Energies & CO2 Manager

***@***.******@***.***>

FAURECIA
Bois sur prés,
25550 Bavans – France
M +33 6 68 97 50 42<tel:+00000000000>

***@***.***

5acXjzUk