SCADA-LTS/Scada-LTS #2780: Display problem for scripts with many context points

nduvernois, 2023-11-21:
I have a script with 20 context points. The list takes some place on the display. Below the list of context points, the script itself is displayed in a box with a dedicated lift. The scripting page (.../scripting.shtm) has a lift also.
By scrolling with the main windows lift, the list of scripts on the left side moves, the central block with context point + script remains steady. If the list of context point is long enough + PC screen is not too large, the script edition area is almost hidden. The only solution is to use the zoom from the navigator, but then font becomes small and difficult to read

Possible solutions 👍
- add a lift within the (context point + script) area ?
- arrange script area on the right part of the window instead of below the list of context point, and use an horizontal lift if necessary ?
- .... ?

 - OS: Windows Server 201610.0
 - Browser: Chrome 118
 - Version: 2.6.18

fabiodurao, 2023-11-21:
Hello, this has already been resolved in version 2.7.5.3 as per the link below.
https://github.com/SCADA-LTS/Scada-LTS/issues/2506

nduvernois, 2023-11-21:
ok. good news then, and sorry for this duplication of a resolved issue

closed 2023-11-21 by nduvernois