SCADA-LTS/Scada-LTS #2788: Bug `npm ERR! Error: Cannot find module 'minimist'`

RenatoExpert, 2023-12-02:
**Describe the bug**
Similar to issue #2505 
I am into branch 'develop'

**To Reproduce**
Steps to reproduce the behavior:
1. Change directory '/scadalts-ui'
2. Install dependencies with NPM 'npm install'
4. See error

**Expected behavior**
It should install NPM modules with success

**Screenshots**
```
┌────(レナトの将軍)-[Scada-LTS]
└─ $ cd scadalts-ui/
┌────(レナトの将軍)-[scadalts-ui]
└─ $ npm i
npm WARN old lockfile 
npm WARN old lockfile The package-lock.json file was created with an old version of npm,
npm WARN old lockfile so supplemental metadata must be fetched from the registry.
npm WARN old lockfile 
npm WARN old lockfile This is a one-time fix-up, please be patient...
npm WARN old lockfile 
npm WARN old lockfile HttpErrorGeneral: 404 Not Found - GET https://registry.npmjs.org/@min-gb%2fvuejs-components - Not found
npm WARN old lockfile     at /usr/lib/node_modules/npm/node_modules/npm-registry-fetch/lib/check-response.js:95:15
npm WARN old lockfile     at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
npm WARN old lockfile     at async RegistryFetcher.packument (/usr/lib/node_modules/npm/node_modules/pacote/lib/registry.js:87:19)
npm WARN old lockfile     at async RegistryFetcher.manifest (/usr/lib/node_modules/npm/node_modules/pacote/lib/registry.js:118:23)
npm WARN old lockfile     at async Array.<anonymous> (/usr/lib/node_modules/npm/node_modules/@npmcli/arborist/lib/arborist/build-ideal-tree.js:727:24)
npm WARN old lockfile  Could not fetch metadata for @min-gb/vuejs-components@0.0.7 HttpErrorGeneral: 404 Not Found - GET https://registry.npmjs.org/@min-gb%2fvuejs-components - Not found
npm WARN old lockfile     at /usr/lib/node_modules/npm/node_modules/npm-registry-fetch/lib/check-response.js:95:15
npm WARN old lockfile     at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
npm WARN old lockfile     at async RegistryFetcher.packument (/usr/lib/node_modules/npm/node_modules/pacote/lib/registry.js:87:19)
npm WARN old lockfile     at async RegistryFetcher.manifest (/usr/lib/node_modules/npm/node_modules/pacote/lib/registry.js:118:23)
npm WARN old lockfile     at async Array.<anonymous> (/usr/lib/node_modules/npm/node_modules/@npmcli/arborist/lib/arborist/build-ideal-tree.js:727:24) {
npm WARN old lockfile   headers: [Object: null prototype] {
npm WARN old lockfile     date: [ 'Sat, 02 Dec 2023 01:27:43 GMT' ],
npm WARN old lockfile     'content-type': [ 'application/json' ],
npm WARN old lockfile     'content-length': [ '21' ],
npm WARN old lockfile     connection: [ 'keep-alive' ],
npm WARN old lockfile     vary: [ 'Accept-Encoding' ],
npm WARN old lockfile     server: [ 'cloudflare' ],
npm WARN old lockfile     'cf-ray': [ '82efc9fb8e9301a1-GRU' ],
npm WARN old lockfile     'x-fetch-attempts': [ '1' ],
npm WARN old lockfile     'x-local-cache-status': [ 'skip' ]
npm WARN old lockfile   },
npm WARN old lockfile   statusCode: 404,
npm WARN old lockfile   code: 'E404',
npm WARN old lockfile   method: 'GET',
npm WARN old lockfile   uri: 'https://registry.npmjs.org/@min-gb%2fvuejs-components',
npm WARN old lockfile   body: { error: 'Not found' },
npm WARN old lockfile   pkgid: '@min-gb/vuejs-components@0.0.7'
npm WARN old lockfile }
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: '@achrinza/node-ipc@9.2.2',
npm WARN EBADENGINE   required: { node: '8 || 10 || 12 || 14 || 16 || 17' },
npm WARN EBADENGINE   current: { node: 'v20.5.1', npm: '10.2.4' }
npm WARN EBADENGINE }
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated mkdirp@0.5.1: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated mkdirp@0.5.1: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated svgo@1.3.2: This SVGO version is no longer supported. Upgrade to v2.x.x.
npm ERR! code 1
npm ERR! path /home/renato/Work/Forks/Scada-LTS/scadalts-ui/node_modules/cypress
npm ERR! command failed
npm ERR! command sh -c node index.js --exec install
npm ERR! node:internal/modules/cjs/loader:1051
npm ERR!   throw err;
npm ERR!   ^
npm ERR! 
npm ERR! Error: Cannot find module 'minimist'
npm ERR! Require stack:
npm ERR! - /home/renato/Work/Forks/Scada-LTS/scadalts-ui/node_modules/cypress/index.js
npm ERR!     at Module._resolveFilename (node:internal/modules/cjs/loader:1048:15)
npm ERR!     at Module._load (node:internal/modules/cjs/loader:901:27)
npm ERR!     at Module.require (node:internal/modules/cjs/loader:1115:19)
npm ERR!     at require (node:internal/modules/helpers:130:18)
npm ERR!     at Object.<anonymous> (/home/renato/Work/Forks/Scada-LTS/scadalts-ui/node_modules/cypress/index.js:3:16)
npm ERR!     at Module._compile (node:internal/modules/cjs/loader:1233:14)
npm ERR!     at Module._extensions..js (node:internal/modules/cjs/loader:1287:10)
npm ERR!     at Module.load (node:internal/modules/cjs/loader:1091:32)
npm ERR!     at Module._load (node:internal/modules/cjs/loader:938:12)
npm ERR!     at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:83:12) {
npm ERR!   code: 'MODULE_NOT_FOUND',
npm ERR!   requireStack: [
npm ERR!     '/home/renato/Work/Forks/Scada-LTS/scadalts-ui/node_modules/cypress/index.js'
npm ERR!   ]
npm ERR! }
npm ERR! 
npm ERR! Node.js v20.5.1

npm ERR! A complete log of this run can be found in: /home/renato/.npm/_logs/2023-12-02T01_27_34_387Z-debug-0.log
```

**Desktop (please complete the following information):**
 - OS: Arch Linux x86_64 - Kernel 6.1.46-1-lts
 - SCADABR Version: 10b3c97506e79c3a3dc761fdf9e74df3d363d7cf
 - Gradle Version: 7.6.3
 - NPM Version: 10.2.4

**Additional context**
I have many difficulties with `docker build` and `gradle war` into project root also.

Limraj, 2023-12-02:
Hi @RenatoExpert,
You are using the wrong version of Node (max 14.20.1) and Gradle(7.1) you will find all information here:
https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Linux

If you customize all versions according to the instructions(the most important thing is that you have compatible versions: node, gradle, and use linux-installer (with java, tomcat, mysql) ), then:

1. git pull/rollback on `.../Scada-LTS`; 
2. Remove: 
  - `.../Scada-LTS/scadalts-ui/node_modules`,
  - `.../Scada-LTS/scadalts-ui/dist`,
  - `.../Scada-LTS/WebContent/resources/node_modules`,
  - `.../Scada-LTS/WebContent/resources/js-ui`;
3. git rollback on `.../Scada-LTS`; 
4. Here is a list of gradle tasks that will allow you to interact with the application's source code:
https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Linux#51-gradle-tasks-

If you use the instruction, the task should work:
`gradle buildRunDebugProd`

Regards,
Kamil Jarmusik

RenatoExpert, 2023-12-04:
Why not upgrading to newer LTS software versions? Those dependencies seems to be too old...

closed 2023-12-04 by RenatoExpert

Limraj, 2023-12-06:
Hi @RenatoExpert,
You are not the first person who would like to do this... :) There is a compatibility problem related to other JS libraries we use, we have to upgrade many versions at the same time, which probably will not have a positive impact on the operation of the new front. Unfortunately, js libraries change so dynamically and backward compatibility is not taken care of, in addition, library dependencies use wildcard marking of the required version, which, given the frequent lack of compatibility between versions, generates new problems, and in the js environment it is a norm to simply remove libraries from a remote repository, today it is there, tomorrow a given version may no longer exist. In general, using modern JS frameworks is a drag, this environment needs a person 24/7 who will keep an eye on it, update and modify sources for new versions of the libraries. If you are brave, you can try to update the versions of node and other libraries, if it works and the application does not lose functionality, we will of course agree to it.

Regards,
Kamil Jarmusik

RenatoExpert, 2023-12-06:
Happy to know its not on purpose!
I am currently learning about Scada-LTS, having hope to contribute at something later.
Regards