SCADA-LTS/Scada-LTS #2794: Traditional graphics are not loading

fabiodurao, 2023-12-13:
**Describe the bug**
Traditional charts are not loading on the datapoint details screen, neither in the graphical representation nor in the watch list when the datapoint has a rendering of the chart render properties (image) type.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the watch list;
2. Choose a numeric datapoint and click the magnifying glass to view the datapoint details;
3. On the datapoint details screen, everything is loaded correctly, except the graph, even forcing the generation of a graph for a defined interval does not work either. The same is true in other places where this type of traditional (old) chart is used.

**Expected behavior**
the graph was expected to load

**Screenshots**
[image]

**Desktop (please complete the following information):**

 - OS: Ubuntu 22.04
 - Browser: Chrome
 - Version: Scada-LTS 2.7.6.1

**Additional context**
In the new UI, amCharts charts in the watch list work.

fabiodurao, 2023-12-14:
Attached are the tomcat logs. For additional context, I asked Fabrício (a github user here who is helping with the Linux installer) how his installation was going, and he told me that these graphics weren't working during his installation either.

[logs.zip](https://github.com/SCADA-LTS/Scada-LTS/files/13666901/logs.zip)

Limraj, 2023-12-14:
[image]

Limraj, 2023-12-14:
[image]

weschenfelder, 2023-12-15:
Hello @fabiodurao &  @Limraj, my projects that are running on Linux also have the same situation, the graphics do not load.

Limraj, 2023-12-16:
@weschenfelder If you don't provide details, there's nothing we can do about it.

Limraj, 2023-12-17:
Hi @fabiodurao and @weschenfelder,
I can't repeat it on Windows and Mac, I still have to check it on Linux.

Regards,
Kamil Jarmusik

fabiodurao, 2023-12-18:
Thanks for the feedback, I have an application on Ubuntu 22.04 with SLTS 2.6.18 and the graph appears normally. I have another application on Windows with SLTS 2.7.5.4 and the graph also appears normally.

In this application that is causing the problem (SLTS 2.7.6.1) on Ubuntu 22.04, I am accessing it via VPN, but I don't believe that is it. I inspected it in Chrome using the F12 button and I didn't find any errors, it may be that I didn't do it correctly, I don't know the resource very well, but I went through all the available tabs and didn't find anything. I've already sent the json privately and the logs, but if you need any testing, you can tell me.

Limraj, 2023-12-18:
On linux mint(base ubuntu)

[image]

after using [installer](https://github.com/SCADA-LTS/linux-installer/releases/download/v1.3.3/linux-installer_v1.3.3_ScadaLTS_v2.7.6.1_Setup.zip) and import data then:
[image]

Brwoser:
1. Open 'http://localhost:8080/Scada-LTS/';
2. Click on F12;
3. Click ctrl+r and ctrl+f5; ('Reload ignore cache')
4. Screenshot from network and console tab;

Tomcat:
5. Stop tomcat;
6. Removed all logs;
7. Start tomcat and try scenario;
8. Send logs;

9. Use feature 'Export project' and send to me zip with config proejct;

fabiodurao, 2023-12-18:
I do step, 5, 6, 7, 1, 2, 3, 4 without click to generate chart.

[image]

[image]

Then I clicked to generate the graph with a period that I'm sure has data

[image]

[image]

I clicked a few more times and this appeared and Tomcat apparently crashed and I tried a reload which kept looping forever.

[image]

[image]

After that I had to stop tomcat and start it again and SLTS came back. I didn't need to restart mysql.

Below are the logs.
Note: I'm using Brave, but I tested it with Chrome and the same thing happened, both are based on Chromium.

[logs.zip](https://github.com/SCADA-LTS/Scada-LTS/files/13709503/logs.zip)

Limraj, 2023-12-19:
I can't open these screenshots, I have a 404...
[image]

fabiodurao, 2023-12-19:
below the prints.

[prints.zip](https://github.com/SCADA-LTS/Scada-LTS/files/13709641/prints.zip)

Limraj, 2023-12-19:
If you want me to help you, please try to follow my instructions, otherwise I am unable to help.

Limraj, 2023-12-19:
An example link, look for a similar link in your home and try to open it in a separate tab:
http://localhost:8080/Scada-LTS/chart/ft_1702940989145_1702854549000_-1_74.png?w=1148&h=350

fabiodurao, 2023-12-19:
For me, all the prints opened in full screen (that's why I downloaded them, compressed them and sent them to you, I thought it was a github problem), but I'm on an ultrawide monitor, I changed the resolution to full HD (1920x1080) to make the prints as asked, maybe it will help.

[image]
[image]

I don't know what's important, so I clicked on the task "ft_1702...." and took screenshots of all the tabs that opened.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

I hope this helps you, if it's not good yet, I'll do it again in whatever way works best.

Limraj, 2023-12-19:
They write here that the cause may be something like adblock: (pending status for the request)
https://stackoverflow.com/questions/5585918/what-does-pending-mean-for-request-in-chrome-developer-window

fabiodurao, 2023-12-19:
I thought it could be that, Brave already has several privacy restrictions and has a native adblocker, but the icon to the left of the red is where the adblock is, and it is disabled, but as a precaution I tested it on Chrome and disabled the adblock plugin and I gave ctrl+f5 and got the same response, and finally, I tested it in Firefox, which is a browser I never use, so it is clean with factory defaults and I got the same response, could it be related to the VPN or to firewall restrictions on the server where the STLS is hosted?

Limraj, 2023-12-19:
@fabiodurao It may be, but I would uninstall ad blocking apps and with that privacy, they are probably causing this problem even if they appear to be disabled.

Limraj, 2023-12-19:
In your log it looks like the request goes to tomcat:
[image]

So I think that here the browser is blocking the content from being displayed.

chrome://settings/content/images?search=site+settings
[image]

chrome://settings/content/insecureContent?search=site+settings
[image]

These may be more advanced programs, or during installation, they configure browsers and even if you turn off this tool, the settings in the browser remain...

fabiodurao, 2023-12-20:
I did all the procedures you mentioned and the behavior remains the same, it doesn't seem to be the browser in general, because I can load the graphic in a locally installed SLTS (2.7.5.4 - it's not the same version, at the moment I can't uninstall this version to test the newest one locally). I uninstalled the adblock plugin, cleared cache, restarted Windows.

[image]

[image]

[image]

There is even a release to display content from http://192.168.0.250:8080/ (SLTS server address). 
I also uninstalled and reinstalled Firefox and it works the same as Chrome, it doesn't load the image.

I don't know how to test "request and response", if you can give me more instructions I will test it.

On this Scada-LTS server (http://192.168.0.250:8080/Scada-LTS) I migrated an application made in ScadaBR 1.2 (address http://192.168.0.156:8080/ScadaBR), this application still it is on the air and has the same VPN restrictions and from what I know it has the same firewall limitations (I don't have access to the restrictions, the IT department only gave me access to the server.
I thought about testing the two applications side-by-side ScadaBR and Scada-LTS and ScadaBR is loading the graphics.

[image]

If there are more tests I can do to contribute, I'm at your disposal.

Limraj, 2023-12-21:
Hi @fabiodurao,
 the only chance to solve the problem is to check all these things and not rely on guesses. So far, there is no indication that the problem is caused by the application (I am not able to repeat it), probably problem is the configuration of the environment, i.e. the server or the client.

1. If in "Insepct->Network" you click on the address from Scada-LTS that generates the chart (the url from the chart), you will see the request and response tab there, you can see it. 
2. You also immediately check the logs that I showed you, those from localhost_access_log, whether a given request has been handled.
3. **The next thing is to check whether you can see these charts on version 2.7.6.1 directly from the server, i.e. on localhost.**
4. **Try to install 2.7.6.1 on another server where the charts work.**
5. Adjust these chrome settings to the state as shown in my screenshot. (all empty)
6. Uninstall Brave, maybe there is something wrong with the configuration of other browsers, or config system. 
7. Do you have any additional antivirus or system protection programs?
8. Check whether the tomcat process on server that runs the application owns all files from CATALINA_HOME.
To set this parameter
`chown -R tomcat_user:user_group dir`

Avoid guessing, test only specific versions, or only on a problematic server, this may only lead to the cause of the problem, the tests you did do not provide specific information, we cannot rely on them.

Limraj, 2024-01-07:
Hi @fabiodurao, @weschenfelder 
How's the situation? Did you check by running tomcat directly from the bin, without using service?

Regards,
Kamil Jarmusik

weschenfelder, 2024-01-10:
good afternoon sirs!

Happy New Year

@Limraj , after testing I identified that the problem occurs when Ubuntu Server is used without the GNOME package, that is, without the Ubuntu graphical interface only through ssh.
If you are interested, I can provide access to the environment with the "problem" so you can debug.

Just provide me with an email and/or Telegram so I can send you the credentials.

Limraj, 2024-01-10:
Hi @weschenfelder,

Thank you and to you too.

Instead of guessing, just try to do what I wrote. 
This is something in common between you and Fabio.

Regards,
Kamil Jarmusik

weschenfelder, 2024-01-10:
O @Limraj 

I'm not good at guessing, so I tested in both environments. In other words, they are not assumptions, they are statements.
In the GNOME version it works.
The problem is in the server version without gnome, which is the most common on servers with few resources.

Regards,
F. Weschenfelder

Limraj, 2024-01-11:
Hi @weschenfelder,
We have this configuration on our servers and there are no such problems, it's just a guess. 
**You have to try what I wrote about.**

Regards,
Kamil Jarmusik

closed 2024-01-30 by Limraj