SCADA-LTS/Scada-LTS #2795: scadalts-ui - sinon is not a dependency

RenatoExpert, 2023-12-17:
I noticed that we could move Sinon from dependencies to DevDependencies, once its only used to make tests.
https://github.com/SCADA-LTS/Scada-LTS/blob/10b3c97506e79c3a3dc761fdf9e74df3d363d7cf/scadalts-ui/package.json#L42

```
[renato@zosoarch scadalts-ui]$ git grep sinon
package-lock.json:              "@sinonjs/commons": {
package-lock.json:                      "resolved": "https://registry.npmjs.org/@sinonjs/commons/-/commons-1.8.3.tgz",
package-lock.json:              "@sinonjs/fake-timers": {
package-lock.json:                      "resolved": "https://registry.npmjs.org/@sinonjs/fake-timers/-/fake-timers-7.1.2.tgz",
package-lock.json:                              "@sinonjs/commons": "^1.7.0"
package-lock.json:              "@sinonjs/samsam": {
package-lock.json:                      "resolved": "https://registry.npmjs.org/@sinonjs/samsam/-/samsam-6.1.1.tgz",
package-lock.json:                              "@sinonjs/commons": "^1.6.0",
package-lock.json:              "@sinonjs/text-encoding": {
package-lock.json:                      "resolved": "https://registry.npmjs.org/@sinonjs/text-encoding/-/text-encoding-0.7.2.tgz",
package-lock.json:                              "@sinonjs/commons": "^1.8.3",
package-lock.json:                              "@sinonjs/fake-timers": ">=5",
package-lock.json:                              "@sinonjs/text-encoding": "^0.7.1",
package-lock.json:              "sinon": {
package-lock.json:                      "resolved": "https://registry.npmjs.org/sinon/-/sinon-11.1.1.tgz",
package-lock.json:                              "@sinonjs/commons": "^1.8.3",
package-lock.json:                              "@sinonjs/fake-timers": "^7.1.0",
package-lock.json:                              "@sinonjs/samsam": "^6.0.2",
package.json:           "sinon": "11.1.1",
tests/unit/Components/Amcharts/AmChart.spec.js:import sinon from "sinon";
tests/unit/Components/Amcharts/AmChart.spec.js:        beforeEach(() => sandbox = sinon.createSandbox());
tests/unit/Components/Amcharts/AmChart.spec.js:            let create = sinon.stub(chart, 'createChart');
tests/unit/Components/Amcharts/AmChart.spec.js:            sinon.assert.calledOnce(create);
tests/unit/Components/Amcharts/AmChart.spec.js:            sinon.assert.calledWith(get, `./api/point_value/getValue/1`);
tests/unit/Components/Amcharts/AmChart.spec.js:            sinon.assert.calledWithMatch(get, requestUrl);
tests/unit/Components/DataPointTables/SimpleTable.spec.js:import sinon from "sinon";
```

Limraj, 2023-12-17:
Hi @RenatoExpert,
what is the purpose of this?

Regards,
Kamil Jarmusik

RenatoExpert, 2023-12-17:
@Limraj only to support #2796 
I will try sending patches on minor granularity

Limraj, 2023-12-17:
@RenatoExpert 
What is this library for?

Limraj, 2023-12-17:
To generate a module's dependency tree and save it to a file:
npm ls --all --unicode -dev > dev.txt
npm ls --all --unicode -prod > prod.txt

1. You mean that sinon is used in prod, and it should be limited to dev, because it is only used by tests? Are you able to prove this using npm commands?

2. Are you sure that this will not cause a problem, or if you make such a change and delete the package-lock.json and node_modules files, will something go wrong? Will some functionality from the new view stop working? This is too small a change and requires a lot of time, tests, etc.

@RenatoExpert  If you want to help us, try upgrading to a higher version of the LTS node.

Regards,
Kamil Jarmusik

closed 2023-12-18 by RenatoExpert