SCADA-LTS/Scada-LTS #2799: Allow modbus tcp slave id to be zero

nichokap, 2024-01-03:
When using some Allen Brandley devices, for example the Powermonitor 3000, and request data over Modbus TCP the device ignores the modbus slave id sent through the requests and replies with the slave id section as zero. Thus, in order to accept the responses you must 

- either be able to set the modbus tcp slave id to 0 when sending the request, or 

- accept that through modbus tcp you can send a request to slave id 1 and get a response from slave id 0.

Limraj, 2024-01-09:
Hi @nichokap,
Could you send us the configuration of this Modbus IP (Data Source and Data Point) in Scada-LTS? We will try to recreate it on our simulator.

Regards,
Kamil Jarmusik

nichokap, 2024-01-09:
Hello @Limraj,

Thanks for looking that quickly the case. Sorry but i didn't understand your request. I cannot apply the configuration needed because when I write slave id 0 I get an error that the slave id should be between 1 and 255. Thus, saving the configuration and creating a data point is not possible. 

If choose modbus slave id, I get a response that no reply received from this slave id (there is a reply actually, but the modbus tcp response packet has replaced slave id with 0). 

Happy to support with anything more you may need.

Limraj, 2024-01-09:
Okej @nichokap, 
I think I now understand what the problem is, you cannot set the slave id to 0, although a physical device is available with such an id, we will try to expand the possibility of entering 0.

nichokap, 2024-01-10:
Thank you once again @Limraj . Sorry if I didn't describe it corrently in the first place!

Patrykb0802, 2024-08-06:
Hi @nichokap,

The issue you're experiencing with slave ID = 0 is because this ID is reserved for broadcast communication in the Modbus protocol. When using Modbus TCP, setting the slave ID to 0 allows a master to send a message to all devices on the network. This means any device receiving a message with slave ID = 0 will process the request but typically will not send a response to avoid network congestion and collisions.

Here’s a reference to this behavior:

[Fernhill Software - Modbus Protocol Overview](https://www.fernhillsoftware.com/help/drivers/modbus/modbus-protocol.html): "A master can broadcast a message to all devices using address 0 (zero)."

I hope this clarifies why the device is responding with slave ID = 0 and helps you understand the reserved usage for broadcast communication.

Best regards,
@Patrykb0802

nichokap, 2024-12-13:
hello @Patrykb0802, thank you for your feedback. The issue I am facing is due to the strange behaviour of this specific Allen Bradley device. The only way to communicate with it is to send Modbus TCP requests with Slave ID =0. The question is from my side is simple, do you still have plans to allow setting Slave ID = 0 in Modbus TCP requests in a future release?

Limraj, 2024-12-14:
Hi @nichokap,
Yes, we will probably add this option in the next version, it does not require much work. However, we are asking for some cooperation, e.g. by committing to confirm whether the correction worked.

Regards.

nichokap, 2024-12-14:
Hi again @Limraj, 
I do commit to check it given that I will still have access to the Allen Bradley devices (as we speak they are  still available on my work environment).

closed 2024-12-18 by Limraj