SCADA-LTS/Scada-LTS #2805: Upgrade Node

Limraj, 2024-01-08:
**Target**
- replacement of js libraries/versions with deprecated status by currently supported versions;
- eliminating vulnerabilities;

**Solution**
- [ ] upgrade Node LTS version (min 20, nice 22, best 24);
- [ ] upgrade versions for other js library (to the minimum version on which the build works);
- [ ] updating supported node versions in the wiki:
https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Linux
https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Windows

**Links**
[Node archive](https://nodejs.org/en/about/previous-releases)
[Upgrade to node 14 - example](https://github.com/SCADA-LTS/Scada-LTS/pull/2366/files)

**Additional context**
Scada-LTS version: 2.7.6.1

YuaFox, 2024-08-22:
After upgrading nodejs it would be nice to migrate vue 2.7 to vue3 and remove some dependences.

For example Scada-LTS uses vuetify.

vue-color-picker-wheel: we can use https://vuetifyjs.com/en/components/color-pickers/#usage
vuejs-datepicker: we can use https://vuetifyjs.com/en/components/date-pickers/#usage

Limraj, 2024-08-22:
Hi @YuaFox,
The purpose of this task is to make the min that are necessary for the project to be built and work. So you have to check it then.
We tend not to remove dependencies, unless you can prove that it is not used by sources and other libraries, but I would suggest doing this in a separate issue.

Regards,
Kamil Jarmusik

closed 2024-10-29 by Limraj