SCADA-LTS/Scada-LTS #2813: Modbus point read illegal data value

nvlong248, 2024-01-25:
Modbus Point locator test returns "Illegal data value" notice while the same register can be read at Modbus read data feature.

[image]

Limraj, 2024-01-25:
Hi @nvlong248,

Does it work with another type?

Regards,
Kamil Jarmusik

nvlong248, 2024-01-26:
Hi @Limraj 
Depends on some registers, it can work with 4 byte integer or 8 byte integer. But it would be helpful if I can add single register as 2 byte integer.
Regards,
Long

Limraj, 2024-06-06:
Hi @nvlong248,
from what I see, these values ​​are 4 bytes, so if we try to read them as 2 bytes, do we have an error? Or maybe I got something wrong? Because if I'm right, this is correct behavior...

Regards,
Kamil Jarmusik

Patrykb0802, 2024-07-09:
Hi @nvlong248,

I looked into this issue however I could not reproduce your error. Below are screenshots of my Scada configuration on different versions: 

version 2.6.18:
[image]

version 2.7.7.3:
[image]

### What version of Scada-LTS application are you using ?
### Could you send screenshot of returned values from Modbus while using Modbus data type 4 or 8 byte unsigned integer on the same configuration ?

Regards,
@Patrykb0802

Patrykb0802, 2024-07-30:
### We tried to reproduce this error, but with no results, if the problem occurs again, please contact us.

**Regards,
Scada-LTS development team**

closed 2024-07-30 by Patrykb0802