SCADA-LTS/Scada-LTS #2815: Update Point Properties Examples

odenisnobre, 2024-01-28:
I'm trying to use the /point_properties/updateProperties endpoint
but it is giving error 400.
I'm passing the cookies in the request and the object in the message:
{"enabled": false}
url: http://192.168.1.82:8088/Scada-LTS/api/point_properties/updateProperties?id=101

My need is to be able to enable or disable a data point via API in order to control when data is collected.

Version: 2.7.6
Platform: Linux
Subsystem: Docker

Limraj, 2024-01-30:
Hi @odenisnobre,
You should have a message in response why you are getting 400. Remember that if you want to change any parameter of a specific point, you must pass the id/xid of that point.

Regards,
Kamil Jarmusik

odenisnobre, 2024-01-30:
Hi @Limraj 
How to use Node-RED are the settings I am using as the message body (on the left) and the request return (right side)

[image]

odenisnobre, 2024-01-30:
Well, using the endpoint you provided, the toggle worked and worked as needed, even if done using Python and not Node-RED for now because PATCH is not found in the current library.

This way it works perfectly.
Now would you be able to get the current status of the tag to know if the toggle will be enabled or disabled?

Limraj, 2024-01-30:
In the response of this endpoint, there is information whether the endpoint has been enabled or disabled

odenisnobre, 2024-01-30:
I understood.
These new endpoints will be interesting for the new version.

Regarding the response to the /toggle request, the status actually came, enable=true/false
But it would be interesting to have visibility of the current status to know if I am enabling or disabling, because only the command and the response to the last command may not work.

Limraj, 2024-01-30:
Try:
GET /api/datapoint?id=point_id

odenisnobre, 2024-01-30:
Perfect.
It works this way.

Thank you very much and this topic can be closed.

odenisnobre, 2024-01-30:
All doubts were answered.
Thanks.