SCADA-LTS/Scada-LTS #2892: Docker database connection conf

Lumevana, 2024-05-23:
I'm trying to run Scada-LTS with docker compose on Ubuntu server.
In yml file I changed database user and passwords because I didnt want to use defaults.
But how I can change database user and password in Scada-LTS?

In documentation is described **context.xml** file change in **/config** directory but I can see **/conf** directory and file content is only three lines about watched resources and no database configuration info.

Currently Tomcat will give 404 and saying that "_The origin server did not find a current representation for the target resource or is not willing to disclose that one exists._"

If I looked into container then the content is there in **Scada-LTS** directory. Is it problem that database user and password is out of sync between app and database that there is no web app error message?

Limraj, 2024-05-23:
Hi @Lumevana,
I have now created a wiki page about creating a new user in the database:
https://github.com/SCADA-LTS/Scada-LTS/wiki/Added-User-MySQL-Database-Server

Does this solve your problem?

Regards,
Kamil Jarmusik

Lumevana, 2024-05-23:
Actually my question is that I didnt find the place where I can enter database user and password in Scada-LTS container.

In my yml file the database section is following:
**database:
        container_name: mysql
        image: mysql/mysql-server:8.0.32
        restart: unless-stopped
        ports:
            - "3306:3306"
        environment: 
            - MYSQL_ROOT_PASSWORD=newrootpassword
            - MYSQL_USER=newuser
            - MYSQL_PASSWORD=newpassword
            - MYSQL_DATABASE=scadalts
        expose: ["3306"]
        volumes:
            - db_data:/var/lib/mysql:rw
            - db_conf:/etc/mysql:ro
        networks: scadanet
        command: --log_bin_trust_function_creators=1**

Now I assume that I must put **newuser** and **newpassword** to context.xml but there is no database section. Only three lines about watched resources. Where I write these new user and password in Scada-LTS?

Limraj, 2024-05-23:
Hi @Lumevana,
First, you need to login to the MySQL server as root, i.e. according to the default configuration, then create a new user according to the instructions. To do this you need a MySQL database client, I recommend DBeaver. After installation and connecting client to the database server from the docker container, you can execute the queries from the instructions.

Steps:
1. Come back to default configuration in docker compose file;
2. Run container with MySQL server; (in console execute command: `docker-compose up database`)
3. Install DBeaver MySQL client and configuration connection client to server MySQL:
[MySQL Client Windows](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Windows#21-mysql-client-)
[MySQL Client Linux](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Linux#21-mysql-client-)
4. Executed queries from instructions;
[Added-User-MySQL-Database-Server](https://github.com/SCADA-LTS/Scada-LTS/wiki/Added-User-MySQL-Database-Server)

These are links to our wiki, so it's worth taking a look there. :)

Regards,
Kamil Jarmusik

Lumevana, 2024-05-24:
Currently the context.xml file looks like this (comments from beginning are excluded):

**<Context>

    
    
    <WatchedResource>WEB-INF/web.xml</WatchedResource>
    <WatchedResource>WEB-INF/tomcat-web.xml</WatchedResource>
    <WatchedResource>${catalina.base}/conf/web.xml</WatchedResource>

    
    
</Context>**

Do I have to manually add section **_<Resource name="jdbc/scadalts" etc_** ?

Limraj, 2024-05-24:
Hi @Lumevana,
Yes. Here is an default file for docker:
https://raw.githubusercontent.com/SCADA-LTS/Scada-LTS/develop/docker/config/context.xml

Regards,
Kamil Jarmusik

Lumevana, 2024-05-24:
OK! 
I was afraid that I made something totally wrong because context.xml file was empty.

Now I added database section and Scada-LTS is up and running and I was able to log in.

One last question. There is classical BR version UI and modern UI. 
Modern UI shows that its offline. I'm running it through Cloudflare tunnel. What ports modern UI is using or what is principally different from classical UI? 

I'm testing it later next to server without Cloudflare to be sure that everything is OK locally.

Limraj, 2024-05-24:
@Lumevana, the greatest functionality is on the old UI, here we focus on developing functions and additionally try to transfer them to the new UI. The new UI is in the experimental phase. I understand that I can close the topic?

Lumevana, 2024-05-24:
Yes, this can be closed. Database is changed and connected.

closed 2024-05-24 by Limraj

Limraj, 2024-05-24:
@Lumevana If you are not familiar with using Docker, we do not recommend using it in production. Better to use the installer.