SCADA-LTS/Scada-LTS #2900: Windows installer v2.0.1 for ScadaLTS 2.7.7.3 not working properly

eng-rodrigo-lobo, 2024-05-27:
**Describe the bug**
I tired to install the new version of ScadaLTS 2.7.7.3 but after the installation process I can't access the Scada UI.
I get a screen with the following text:

System exception!

The server has experienced an exception processing your last request. The exception has been logged, and system administrators will be notified of this problem. You may continue to use the site. We apologize for the inconvenience.

Please provide the following code to the administrator to identify the problem:
20240527182954591

**To Reproduce**
Steps to reproduce the behavior:
1. Open the Windows Installer v2.0.1
2. Proceed with the installation
3. Try to open ScadaLTS for the fisrt time after the instalation is completed
4. See error

**Expected behavior**
I expected to access the ScadaLTS UI to program the system, add datasources, etc

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Microsoft Edge
 - Version: 125.0.2535.67

**Additional context**
Tomcat seems to be running fine. I can access the server, start and stop, no big deal. I noticed that no tables were created after the installation. I dumped a bkp after the installation, so the expected tables would be available, but the problem still persists.

I checked the logs and it looks it's getting some trouble when starting the services, but it's too advanced for my knowledge... I can provide them if necessary.

[scada-lts-stdout.2024-05-27.log](https://github.com/SCADA-LTS/Scada-LTS/files/15461975/scada-lts-stdout.2024-05-27.log)
[catalina.2024-05-27.log](https://github.com/SCADA-LTS/Scada-LTS/files/15461977/catalina.2024-05-27.log)
[commons-daemon.2024-05-27.log](https://github.com/SCADA-LTS/Scada-LTS/files/15461978/commons-daemon.2024-05-27.log)
[localhost.2024-05-27.log](https://github.com/SCADA-LTS/Scada-LTS/files/15461979/localhost.2024-05-27.log)
[localhost_access_log.2024-05-27.txt](https://github.com/SCADA-LTS/Scada-LTS/files/15461980/localhost_access_log.2024-05-27.txt)
[scada-lts-stderr.2024-05-27.log](https://github.com/SCADA-LTS/Scada-LTS/files/15461981/scada-lts-stderr.2024-05-27.log)

eng-rodrigo-lobo, 2024-05-28:
Hey @Limraj !

I tried your suggestions and now it's working fine!

To be honest I'm not sure what was the actual problem, but I suspect it's related to the environmento variable, because after I created it the application started working fine.

Is it possible to embed this procedure into the installer? So we don't have to do it manualy every time we install the program.

Limraj, 2024-05-30:
Corrected installator: https://github.com/SCADA-LTS/windows-installer/releases/tag/v2.0.2

closed 2024-05-30 by Limraj

Limraj, 2025-01-30:
Hi @Kaushik0o7,
Can you provide more details?

Regards,
Kamil Jarmusik