SCADA-LTS/Scada-LTS #2903: Page not found error

Lumevana, 2024-05-29:
Suddenly Scada-LTS stopped working and displays "Page Not Found" error.
It worked and no changes were made.
Login screen is OK but after login a weird page is displayed (dev tools window is also opened):

[image]

MySQL container is OK, database is accessible, scadalts database exists and tables contain data. ScadaLTS container is running and Scada-LTS directory in place.

Seems like views.shtm is not found. Where it is and what to check? 
I will not find views.shtm file in Scada-LTS directory and subdirectories.

Limraj, 2024-05-30:
Hi @Lumevana,

No wonder you have a no found page, we have nothing at 
`.../Scada-LTS/S/views.shtm`,
the correct address is 
`.../Scada-LTS/views.shtm`

I wonder where that 'S' came from...

Regards,
Kamil Jarmusik

Lumevana, 2024-05-30:
I created new user and made few tests with setting user default homepage.
If default homepage is not set then everything is OK.
After setting homepage, then logout and then after login there is S in URL path.

Default home page is set to "Gaphical views". But same error when setting it to "Watch list".

Lumevana, 2024-05-30:
In database the users table homeURL contains wrong URL:
[image]

I was not able to change it with SQL query page. I will change it later directly in database and see what happens.

Limraj, 2024-05-30:
Hi @Lumevana,
How did you set the Homepage for the user? 
What will happen if you try to overwrite the url without S in user?

Lumevana, 2024-05-31:
I added it via button "Make this page to my homepage" on top bar.

If I fix it via UI then its OK. 

I will play and try to figure out when it adds S in front of URL.

Lumevana, 2024-06-03:
Yes, I found the combination that adds S/ prefix to URL.

I'm using Cloudflare and feature called Zero trust tunnel for connecting to web-based services in private network.
If I access Scada-LTS instance through this tunnel (through public hostname what is assigned to it), then URL will be saved with S/ prefix.
Also, modern UI will not work through Cloudflare tunnel. I haven't found time yet to figure out what is restricting modern UI.
Modern UI works in private network when I access it with private IP.

Limraj, 2024-06-03:
@Lumevana 
So this is not an application problem and this topic can be closed? Problem with network configuration?

Lumevana, 2024-06-03:
It's rather combination what causes problem: application and network access.
Usually web applications are working without problems with Cloudflare tunnels. Scada-LTS using something differently. Do you have any clue what is different between classical and modern UI network usage?

Limraj, 2024-06-03:
@Lumevana
Our application is used by the client in the VPN tunnel and the old and new UI works.
However, you should know that the new user interface is in the experimental phase, it has limited functionality compared to the old user interface, so we recommend using the old user interface, which is why the user after login goes to the old ui and not the new ui. If the new UI doesn't work for you only with Cloudflare, or it adds some words/characters to the URL, then it means it's not an application problem but a network configuration problem, am I wrong? If I had any knowledge of Cloudflare I would try to help, but I have no knowledge of it. This prefix problem is really strange, but we have finite resources and we can't take care of everything and investigate everything thoroughly, such as specific user network configurations.

However, you can write exactly what version Scada-LTS/tomcat/system you are using and what is happening. It will always be some information.

Lumevana, 2024-06-03:
It's runs on ubuntu server as docker container.
v2.7.7.3 build 9060221738 (GitHub ref: 4efbbb2);
runs on Linux5.15.0-105-generic

Limraj, 2024-06-03:
@Lumevana You are not even able to provide specific information that would allow us to refer and investigate anything. I'm closing the topic, I hope you will be able to solve the problem.

closed 2024-06-03 by Limraj