SCADA-LTS/Scada-LTS #2918: report is not sent if the number of datapoints included is higher than 255 (exact value TBC)

nduvernois, 2024-06-07:
Create about 260 datapoints with enough history. Create a report that tries to extract data for a given timeframe on these variables, that is sent by e-mail.

Bug : the e-mail is not sent. I have not searched the exact limit value, but with 243 variables it works, and with about 20 more, the e-mail is not sent. I assume that the threshold is 255 because this is it the maximum value for 1 byte.

Expacted behaviour : either there is no limit, or there is a protection in the report edition window that prevents to overcome the limit.

OS : Windows Server 2016
Browser : Chrome 125
Version : ScadaLTS 2.6.18

nduvernois, 2024-07-01:
update : it doesn't work with 253 datapoints, 251 and 249.  But it works with 247.

Patrykb0802, 2024-07-10:
Hi @nduvernois,

I tried to reproduce your error however without success. I created 254 points that generate binary values every second, then I created a report that contains these points, in the report I had 7620 values generated and the email was sent and received. I tested such a scenario on two versions. Below are screenshots of the tests:

### Version 2.6.18 build 3789793770:
[image]

### Version 2.7.7.3 build 3789793774:
[image]

**We observed that when Scada was installed on the HDD drive it could take more than a minute to generate reports, and sending as well as receiving an email was delayed in relation to report generation by 30 seconds or so. Reinstalling Scada to an SSD drive reduced the report generation time to a few seconds and eliminated the problem of delayed report sending.**

Regards,
@Patrykb0802

nduvernois, 2024-07-10:
Hello @Patrykb0802 
That's weird. Looking back in report history, the initial report with 253 datapoints (1 value/datapoint) was launched & executed but not sent on 1st of July at 00:10. I realised it later in the day, and, as described in my previous comment, I had to reduce step by step the number of datapoints to make it work. 2 other reports with a very similar list of datapoints (same number of datapoints, but with 1 month history and 1,1 to 1,4 M values) showed the same behaviour. 
So ... I don't know.
By the way, my ScadaLTS installation is on SSD drives since the beginning.
Regards
Nicolas

Patrykb0802, 2024-07-10:
Could you force generation of the same report with 254 data points that was not emailed on July 1 ? If the email is not sent then send us logs of the time when the report generation occurred.

Regrads,
@Patrykb0802

closed 2024-12-05 by Limraj