SCADA-LTS/Scada-LTS #2920: MQTT add new datasource can't save

subairi, 2024-06-09:
**Describe the bug**
i just install from [Scada-LTS_v2.7.7.3_Installer_v2.0.2_Setup.exe] , when add Data Sources MQTT and click save button, but can't save, just blink icon save for long time.
server host: test.mosquitto.org
port: 1883

Limraj, 2024-06-10:
Hi @subairi,
[image]
I cannot reproduce this problem, I run the mqtt broker from Docker and I can save this data source normally, after almost 2 seconds the information data source saved appears.

1. In 'Current alarms' there are also entries when the broker was not running. Do you see any errors in 'Current alarms'? You can send me a screenshot.
2. Send tomcat/logs, Do you see any errors in the logs?
3. Try changing the port or host to an address where there is no broker and how will it behave then? Do you see any errors in 'Current alarms'?
4. Try changing the Mqtt protocol versions:
[image]

Regards,
Kamil Jarmusik

subairi, 2024-06-11:
**************************************************
*             Welcome to Scada-LTS!              *
**************************************************
* Version: v2.7.7.3
* Build: 9060221738
* Powered by: Scada-LTS team
 
ERROR 2024-06-11T08:37:13,020 (org.scada_lts.mango.service.SystemSettingsService.getWebResourceGraphicsPath:433) - PreparedStatementCallback; bad SQL grammar [select settingValue from systemSettings where settingName=? ]; nested exception is com.mysql.jdbc.exceptions.jdbc4.MySQLSyntaxErrorException: Table 'scadalts.systemsettings' doesn't exist 
INFO  2024-06-11T08:37:15,357 (com.serotonin.mango.db.MySQLAccess.newDatabaseCheck:103) - schemaExist:false 
INFO  2024-06-11T08:37:15,357 (com.serotonin.mango.db.MySQLAccess.newDatabaseCheck:111) - BaseLineNotExist:true 
WARN  2024-06-11T08:37:15,373 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.warn:53) - Use of dots (.) as path separators will be deprecated in Flyway 7. Path: org.scada_lts.dao.migration.mysql 
INFO  2024-06-11T08:37:15,388 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Flyway Community Edition 6.5.5 by Redgate 
INFO  2024-06-11T08:37:15,451 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Database: jdbc:mysql://localhost:3306/scadalts (MySQL 8.0) 
INFO  2024-06-11T08:37:15,542 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Successfully validated 30 migrations (execution time 00:00.030s) 
INFO  2024-06-11T08:37:15,570 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Creating Schema History table `scadalts`.`schema_version` ... 
INFO  2024-06-11T08:37:15,653 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Current version of schema `scadalts`: << Empty Schema >> 
INFO  2024-06-11T08:37:15,653 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 1 - BaseVersion 
INFO  2024-06-11T08:37:17,826 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 1.1 - ViewsHierarchy 
INFO  2024-06-11T08:37:17,919 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 1.2 - SetViewSizeProperties 
INFO  2024-06-11T08:37:17,935 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 1.3 - SetXidPointHierarchy 
INFO  2024-06-11T08:37:17,982 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.0 - CMP history 
INFO  2024-06-11T08:37:18,015 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.3 - FaultsAndAlarms 
WARN  2024-06-11T08:37:18,031 (org.scada_lts.dao.migration.mysql.V2_3__FaultsAndAlarms.migrate:76) - org.springframework.jdbc.BadSqlGrammarException: StatementCallback; bad SQL grammar [ALTER TABLE dataPoints DROP COLUMN pointName;]; nested exception is com.mysql.jdbc.exceptions.jdbc4.MySQLSyntaxErrorException: Can't DROP 'pointName'; check that column/key exists 
WARN  2024-06-11T08:37:18,037 (org.scada_lts.dao.migration.mysql.V2_3__FaultsAndAlarms.migrate:92) - org.springframework.jdbc.BadSqlGrammarException: StatementCallback; bad SQL grammar [ALTER TABLE dataPoints DROP COLUMN plcAlarmLevel;]; nested exception is com.mysql.jdbc.exceptions.jdbc4.MySQLSyntaxErrorException: Can't DROP 'plcAlarmLevel'; check that column/key exists 
INFO  2024-06-11T08:37:18,112 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.4 
INFO  2024-06-11T08:37:18,138 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.5 - ScheduledExecuteInactiveEvent 
INFO  2024-06-11T08:37:18,200 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.6 
INFO  2024-06-11T08:37:18,231 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.6.5.0 - ZIndexForViewComponent 
INFO  2024-06-11T08:37:18,247 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.7.0.1 - UserParameters 
INFO  2024-06-11T08:37:18,294 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.7.0.2 - FixViewPermissions 
INFO  2024-06-11T08:37:18,310 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.7.0.3 - AnonymousUser 
INFO  2024-06-11T08:37:18,325 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.7.0.4 - ExtendedDelayForMetaDatapoints 
INFO  2024-06-11T08:37:18,341 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.7.0.4.1 - CorrectProcedurePrcAlarmsNotify 
INFO  2024-06-11T08:37:18,356 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.7.0.4.2 - MultiChangeHistory 
INFO  2024-06-11T08:37:18,388 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.7.0.5 - SynopticPanel 
INFO  2024-06-11T08:37:18,403 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.7.1.0 - UserNames 
INFO  2024-06-11T08:37:18,450 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.info:49) - Migrating schema `scadalts` to version 2.7.1.1 - HttpRetriever 
[... cut ...] sVolatile: false calendar: 'null' misfireInstruction: 0 nextFireTime: Tue Jun 11 08:37:07 ICT 2024 
INFO  2024-06-11T08:37:18,873 (com.serotonin.mango.MangoContextListener.initSchedule:655) - Quartz EverySecond initialized 
INFO  2024-06-11T08:37:18,873 (com.serotonin.mango.MangoContextListener.initialized:197) - Scada-LTS context started 
INFO  2024-06-11T10:00:00,300 (org.scada_lts.quartz.ResetCacheHighestAlarmLevel.execute:57) - ResetCacheHighestAlarmLevel executed in [141] ms 
Stable Library

subairi, 2024-06-11:
this is major error?

ERROR 2024-06-11T08:37:13,020 (org.scada_lts.mango.service.SystemSettingsService.getWebResourceGraphicsPath:433) - PreparedStatementCallback; bad SQL grammar [select settingValue from systemSettings where settingName=? ]; nested exception is com.mysql.jdbc.exceptions.jdbc4.MySQLSyntaxErrorException: Table 'scadalts.systemsettings' doesn't exist

Limraj, 2024-06-11:
@subairi Add logs as files from tomcat/logs.

subairi, 2024-06-11:
[image]

[image]

* All value success 1

subairi, 2024-06-11:
[api.log](https://github.com/user-attachments/files/15785230/api.log)
[catalina.2024-06-11.log](https://github.com/user-attachments/files/15785231/catalina.2024-06-11.log)
[commons-daemon.2024-06-11.log](https://github.com/user-attachments/files/15785232/commons-daemon.2024-06-11.log)
[dao.log](https://github.com/user-attachments/files/15785233/dao.log)
[flyway.log](https://github.com/user-attachments/files/15785234/flyway.log)
[history.log](https://github.com/user-attachments/files/15785235/history.log)
[host-manager.2024-06-11.log](https://github.com/user-attachments/files/15785236/host-manager.2024-06-11.log)
[localhost.2024-06-11.log](https://github.com/user-attachments/files/15785237/localhost.2024-06-11.log)
[manager.2024-06-11.log](https://github.com/user-attachments/files/15785238/manager.2024-06-11.log)
[mango.log](https://github.com/user-attachments/files/15785239/mango.log)
[modbus.log](https://github.com/user-attachments/files/15785240/modbus.log)
[scadaLTSDAO.log](https://github.com/user-attachments/files/15785241/scadaLTSDAO.log)
[scada-lts-stderr.2024-06-11.log](https://github.com/user-attachments/files/15785242/scada-lts-stderr.2024-06-11.log)
[scada-lts-stdout.2024-06-11.log](https://github.com/user-attachments/files/15785243/scada-lts-stdout.2024-06-11.log)
[scripts.log](https://github.com/user-attachments/files/15785244/scripts.log)
[viehierarchyservicelogfile.log](https://github.com/user-attachments/files/15785245/viehierarchyservicelogfile.log)

subairi, 2024-06-11:
[image]

I was click save for 2 minutes but nothing happen

Limraj, 2024-06-11:
@subairi, 
1. In addition to the attempt to create one, have you created another Data Source?
2. Try changing the Mqtt protocol versions ?
3. I guess it can't connect to the broker and is waiting for timeout. You can reduce the timeout (e.g. to 1000), if it somehow affects the behavior ?
4. Try set localhost?

subairi, 2024-06-13:
[image]

i Install version 2.7.6.1 is running well, 
while I use this version, thank you

closed 2024-06-13 by Limraj

subairi, 2024-06-16:
1. i install version 2.7.6.1 it was running, then i change SCADA_LTS.WAR with version 2.7.7.3 
   - first error on running tomcat 
  
[localhost_access_log.2024-06-16.txt](https://github.com/user-attachments/files/15855536/localhost_access_log.2024-06-16.txt)
[catalina.2024-06-16.log](https://github.com/user-attachments/files/15855537/catalina.2024-06-16.log)
[localhost.2024-06-16.log](https://github.com/user-attachments/files/15855538/localhost.2024-06-16.log)
[mango.log](https://github.com/user-attachments/files/15855539/mango.log)
[scada-lts-stderr.2024-06-15.log](https://github.com/user-attachments/files/15855540/scada-lts-stderr.2024-06-15.log)
[scada-lts-stdout.2024-06-15.log](https://github.com/user-attachments/files/15855541/scada-lts-stdout.2024-06-15.log)

2- i remove some text on Server.xml 
[image]

3. restart tomcat and running well version 2.7.7.3
4. but when i make new datasource MQTT there is stag on save new datasource proses save, 
in tomcat cli any warning like this:

INFO  2024-06-16T10:19:45,258 (org.scada_lts.quartz.ResetCacheHighestAlarmLevel.execute:57) - ResetCacheHighestAlarmLevel executed in [31] ms
16-Jun-2024 10:20:02.467 SEVERE [MQTT Ping: 0a8539e4-4034-48b7-ac3f-bb01a8ecb6ed] org.eclipse.paho.mqttv5.client.internal.ClientState.checkForActivity 0a8539e4-4034-48b7-ac3f-bb01a8ecb6ed: Timed out as no write activity, keepAlive=20.000.000.000 lastOutboundActivity=32.946.731.940.799 lastInboundActivity=48.962.325.931.000 time=48.979.906.063.600 lastPing=32.946.731.944.699
16-Jun-2024 10:21:15.174 WARNING [MQTT Call: 0a8539e4-4034-48b7-ac3f-bb01a8ecb6ed] org.apache.tomcat.jdbc.pool.interceptor.SlowQueryReport.reportSlowQuery Slow Query Report SQL=insert pointValues (dataPointId,dataType,pointValue,ts) values (?,?,?,?); time=2110 ms;

subairi, 2024-06-16:
[scada LTS exe CLI record.txt](https://github.com/user-attachments/files/15856080/scada.LTS.exe.CLI.record.txt)

subairi, 2024-06-16:
In addition to the attempt to create mqtt Data Source, have you created other Data Source?

other data source running well

Try changing the Mqtt protocol versions ?

I was try changing protocol version 3 and 5, but nothing happen

I guess it can't connect to the broker and is waiting for timeout. You can reduce the timeout (e.g. to 1000), if it somehow affects the behavior ?

this condition was try to change timeout to 3000
Try set localhost? Does it behave the same with any host? this was try but nothing/ still probelm

subairi, 2024-06-16:
2024-06-16 11:56:31 Apache Commons Daemon procrun stderr initialized.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.eclipse.paho.mqttv5.client.internal.FileLock (file:/C:/Program%20Files/Scada-LTS/tomcat/webapps/Scada-LTS/WEB-INF/lib/org.eclipse.paho.mqttv5.client-1.2.5.jar) to method sun.nio.ch.FileLockImpl.release()
WARNING: Please consider reporting this to the maintainers of org.eclipse.paho.mqttv5.client.internal.FileLock
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

Limraj, 2024-06-16:
@subairi, 
I'm sorry, but I can't help you if you do such stupid things instead of following the instructions and describing the situation. It's just not possible if you have these ideas. 
If you change your attitude and really need help, please contact me.

subairi, 2024-06-20:
I was installed this version https://github.com/SCADA-LTS/windows-installer/releases/tag/v2.0.2
that problem on browser (EDGE), i change to use opera browser and it's running well,
thanks for your attention