SCADA-LTS/Scada-LTS #2974: Synoptic panel state not loading correctly

YuaFox, 2024-07-31:
**Describe the bug**
I have the following state:
[image]

If im in the panel and i change the value to 0, the state color is red.
When i refresh the window it shows as green, it should be red.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the panel
2. Chnage the value from 1->0
3. The color is changed to red
4. Refresh
5. (Bug) The color is green, should be red

**Expected behavior**
State should evaluate the condition when loading the page but it always marks it as positive (green color).

**Screenshots**
State when value changes to 0
[image]
After refreshing the tab
[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser Chrome 126.0.6478.182 (64-bit)

YuaFox, 2024-08-01:
When the synoptic panel is loaded onPointEnabledUpdate and onPointValueUpdate are executed.
[image]

```
onPointEnabledUpdate(enabled) {
    if(enabled) {
       ....
        // This should check what color to use
        this.changeComponentColor(`${this.componentId}`, this.enabledColor); 
    } else {
        ....
    }
},```

There's a race between the 2 functions:
- onPointEnabledUpdate always use the enabled color
- onPointValueUpdate checks the condition before using enabled/disabled color

onPointEnabledUpdate should check what color to use when enabling

Limraj, 2024-08-01:
Hi @YuaFox,
Synoptic panels are an interesting solution and are worth investing in. Thanks you.

Propose issue:
#2978 Added ability set value on Synoptic Panel for Data Point settable;

Other issue suggestions:
#2805 Upgrade Node;
#2976 Fixed refresh chart-line and chart-range;

**However, if these issues do not suit your preferences, we are open to your further suggestions for improvements.**

Regards,
Kamil Jarmusik

Limraj, 2024-08-05:
Hi @YuaFox,
we have two more suggestions:
#2955
#2957

Regards,
Kamil Jarmusik

YuaFox, 2024-08-05:
Hi @Limraj 

These suggestions are interesting and will improve the quality of the code, I will review them and see what I can do.

Another improvement that would be interesting would be some kind of interactivity, e.g. actuation buttons. Combined with node-red many interesting things could be done.

Node-red is a widely used plcs/iot automation software.

https://nodered.org/

Limraj, 2024-08-05:
@YuaFox, Unfortunately, the constant introduction of revolutions has led us to a place where the new UI is a burden for us. Rather, we try to improve what is there by minimizing the number of changes. However, you can create an issue on this topic, describe how we can use it, with links to documentation, it will be something for us to consider.

Limraj, 2024-08-05:
https://github.com/SCADA-LTS/Scada-LTS/pull/2979
Here you still have something to improve, it is worth trying to finish one issue before starting the next one. Unless the task is too complex, perhaps poorly defined, too general, etc. Then you can also propose an new issue for a given issue, with a more complete description and a realistic scope of work.

closed 2024-08-24 by Limraj