SCADA-LTS/Scada-LTS #2993: System exception!

luwz, 2024-08-26:
System exception!

The server has experienced an exception processing your last request. The exception has been logged, and system administrators will be notified of this problem. You may continue to use the site. We apologize for the inconvenience.

Please provide the following code to the administrator to identify the problem:
20240826100839165

Limraj, 2024-08-26:
Hi @luwz,
If you do not describe the circumstances of the error, we will not be able to do anything:
1. What operating system are you using?
2. What version of the application/branch are you using?
3. What steps did you take to build your war?
4. Have you followed the instructions in the manual?
[Getting-started-Development-Linux](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Linux)
[Getting-started-Development-Windows](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Windows)

5. Upload logs from tomcat/Intellija;

Regards,
Kamil Jarmusik

Limraj, 2024-08-28:
Hi @luwz,
We invite you to ask questions and solve problems with environment configuration here:
https://github.com/SCADA-LTS/Scada-LTS/discussions

Regards,
Kamil Jarmusik

closed 2024-08-28 by Limraj