SCADA-LTS/Scada-LTS #2994: May I ask why I can use the official Scada LTS.war image package, but when I use IDEA to create my own Scada LTS.war package locally, it doesn't work properly?

luwz, 2024-08-26:
May I ask why I can use the official Scada LTS.war image package, but when I use IDEA to create my own Scada LTS.war package locally, it doesn't work properly?

Limraj, 2024-08-26:
Hi @luwz,
you haven't written anything that could lead us to the cause of your problems. 
Try to install the environment according to the instructions:
https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Windows
https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started-Development-%E2%80%90-Linux

Regards,
Kamil Jarmusik

Limraj, 2024-08-28:
Hi @luwz,
We invite you to ask questions and solve problems with environment configuration here:
https://github.com/SCADA-LTS/Scada-LTS/discussions

Regards,
Kamil Jarmusik

closed 2024-08-28 by Limraj