SCADA-LTS/Scada-LTS #3002: Application independent of the CATALINA_HOME system variable

Limraj, 2024-08-29:
**Description**
If the CATALINA_HOME system variable is not set, the application crashes.

**To Reproduce**
1. Install Scada-LTS by installator: https://github.com/SCADA-LTS/windows-installer/releases/tag/v2.0.3
2. Remove CATALINA_HOME system variable then start tomcat with application -> error:
[image]

**Spec:**
 - Scada-LTS Version: <= 2.7.8

fabiodurao, 2024-08-30:
I found a bug in the default installation and it is probably related to what was mentioned above.

1 - I installed the indicated Java but did not set it as the default Java for my Windows (I always did it this way without leaving JAVA_HOME configured for the version that Scada-LTS uses);
2 - When installing Scada-LTS I pointed to the installation path of the indicated Java that I had just installed;
3 - After completing the installation the application did not work.

I started investigating and realized that not even the MySQL tables were created. Also, in the Tomcat service manager (accessing the service manager with the desktop icon) in the "Java" tab, the path to the "Java Virtual Machine" was not respecting what I had indicated in the installation. It was pointing to a Java 8 that was probably the JAVA_HOME of my Windows. I stopped Tomcat, changed the address to "C:\Program Files\Microsoft\jdk-11.0.18.10-hotspot\bin\server \jvm.dll" and started it. The tables were created and everything started working correctly.

This problem did not exist in previous versions of the installer. In this latest version, several colleagues and I had the same problem, and this workaround worked for everyone.

Limraj, 2024-08-30:
Hi @fabiodurao, 
In fact, in the installer we have a choice of Java, and it probably chooses the one set in the system in JAVA_HOME.
However, the requirement to set JAVA_HOME is described in the manual (our wiki) and and there is no need for such miracles. It has worked this way from the beginning.

Regards,
Kamil Jarmusik

fabiodurao, 2024-08-31:
Thanks for the feedback,

I was going to do exactly that, report it in the discussion, I just took advantage of the fact that I saw the bug and thought I could contribute to the topic.

I saw that the installer's readme says to define JAVA_HOME during installation and this shouldn't be a problem for professional applications, because most of the time Scada-LTS will be installed in isolation on a machine, but on the PCs of new users who are still learning how to use Scada-LTS, this might be a problem, because they might use other Java applications that are set to work with another version of Java, which was defined in the Windows environment variable (JAVA_HOME). If the installer could maintain the standard of indicating the Java version that will be used for Scada-LTS (as it was in the past), I believe it would be better for everyone.

Limraj, 2024-08-31:
Fabio, the instructions say that you set JAVA_HOME when installing Microsoft's jdk, you know that it was not us who prepared this installer, it was Microsoft. You are informed that this variable must be set in the system, there is no way to set the JAVA_HOME variable in our installer, because we do not install Java in it, it has to be done already. What you write here has nothing to do with this topic. 

I repeat again, there is no such thing as installing multiple java versions on one windows account, do you understand that?
Installed java in the windows system means that either we have the JAVA_HOME variable set and we can set it per system account/user, or the java program has been moved to system32 and there is one java for all system users.

Limraj, 2024-09-02:
Hi @fabiodurao,
In the next version of the installer, we will improve this Java setting so that the generated files and shortcuts are independent of JAVA_HOME from env windows, so that you can set any folder with java.

Regards,
Kamil Jarmusik

closed 2024-09-03 by Limraj