SCADA-LTS/Scada-LTS #3012: Alarm List widget not working

nex2000, 2024-09-14:
Hello

I inserted the Alarm list component into a graphic page, even though there are alarms, it does not signal anything.
The release is 2.7.8

[image]

[image]

Regards

nex2000, 2024-09-15:
it worked now but i had to restart the server

Regards

Limraj, 2024-10-23:
Hi @nex2000,
We cannot reproduce this, perhaps a more restrictive alert level was saved and then changed in select but this change was not saved using the "Save" button. 
Another possible cause is that the UI Performance parameter in System Settings is set to refresh too rarely, because the table has not been updated with events for some time.

Regards,
Kamil Jarmusik

closed 2024-10-23 by Limraj