SCADA-LTS/Scada-LTS #3024: Database migrations not working properly (new columns not added)

felipecarneiro, 2024-09-26:
I just did a fresh install of ScadaLTS v2.7.8 on a new Linux server using the .war file, it creates the database in the first run, but there are some new fields that are not present.

This seems to be related to: https://github.com/SCADA-LTS/Scada-LTS/issues/1947#issuecomment-972915101

These errors are shown in tomcat logs:
```

26-Sep-2024 02:14:33.473 SEVERE [main] org.apache.catalina.core.ApplicationContext.log Servlet.init() for servlet [springDispatcher] threw exception
	org.springframework.jdbc.BadSqlGrammarException: PreparedStatementCallback; bad SQL grammar [select e.id, e.typeId, e.typeRef1, e.typeRef2,e.activeTs,e.rtnApplicable, e.rtnTs,e.rtnCause, e.alarmLevel, e.message, e.shortMessage, e.ackTs, e.ackUserId, u.username,e.alternateAckSource from events e left join users u on e.ackUserId=u.id  where  e.rtnApplicable=? and (e.rtnTs is null or e.rtnTs=0)]; nested exception is com.mysql.jdbc.exceptions.jdbc4.MySQLSyntaxErrorException: Unknown column 'e.shortMessage' in 'field list'
		at org.springframework.jdbc.support.SQLErrorCodeSQLExceptionTranslator.doTranslate(SQLErrorCodeSQLExceptionTranslator.java:231)
		at org.springframework.jdbc.support.AbstractFallbackSQLExceptionTranslator.translate(AbstractFallbackSQLExceptionTranslator.java:73)
		at org.springframework.jdbc.core.JdbcTemplate.execute(JdbcTemplate.java:655)
		at org.springframework.jdbc.core.JdbcTemplate.query(JdbcTemplate.java:690)
		at org.springframework.jdbc.core.JdbcTemplate.query(JdbcTemplate.java:722)
		at org.springframework.jdbc.core.JdbcTemplate.query(JdbcTemplate.java:732)
		at org.springframework.jdbc.core.JdbcTemplate.query(JdbcTemplate.java:782)
		at org.scada_lts.dao.event.EventDAO.filtered(EventDAO.java:900)
		at org.scada_lts.mango.service.EventService.getActiveEvents(EventService.java:198)
		at com.serotonin.mango.rt.EventManager.initialize(EventManager.java:324)
		at com.serotonin.mango.MangoContextListener.eventManagerInitialize(MangoContextListener.java:500)
		at com.serotonin.mango.MangoContextListener.initialized(MangoContextListener.java:143)
		at com.serotonin.mango.MangoContextListener.contextInitialized(MangoContextListener.java:93)
		at org.scada_lts.web.beans.ScadaContextRefreshedEvent.onApplicationEvent(ScadaContextRefreshedEvent.java:24)
		at org.scada_lts.web.beans.ScadaContextRefreshedEvent.onApplicationEvent(ScadaContextRefreshedEvent.java:11)
		at org.springframework.context.event.SimpleApplicationEventMulticaster.doInvokeListener(SimpleApplicationEventMulticaster.java:172)
		at org.springframework.context.event.SimpleApplicationEventMulticaster.invokeListener(SimpleApplicationEventMulticaster.java:165)
		at org.springframework.context.event.SimpleApplicationEventMulticaster.multicastEvent(SimpleApplicationEventMulticaster.java:139)
		at org.springframework.context.support.AbstractApplicationContext.publishEvent(AbstractApplicationContext.java:393)
		at org.springframework.context.support.AbstractApplicationContext.publishEvent(AbstractApplicationContext.java:347)
		at org.springframework.context.support.AbstractApplicationContext.finishRefresh(AbstractApplicationContext.java:883)
		at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:546)
		at org.springframework.web.servlet.FrameworkServlet.configureAndRefreshWebApplicationContext(FrameworkServlet.java:668)
		at org.springframework.web.servlet.FrameworkServlet.createWebApplicationContext(FrameworkServlet.java:634)
		at org.springframework.web.servlet.FrameworkServlet.createWebApplicationContext(FrameworkServlet.java:682)
		at org.springframework.web.servlet.FrameworkServlet.initWebApplicationContext(FrameworkServlet.java:553)
		at org.springframework.web.servlet.FrameworkServlet.initServletBean(FrameworkServlet.java:494)
		at org.springframework.web.servlet.HttpServletBean.init(HttpServletBean.java:171)
		at javax.servlet.GenericServlet.init(GenericServlet.java:143)
		at org.apache.catalina.core.StandardWrapper.initServlet(StandardWrapper.java:984)
		at org.apache.catalina.core.StandardWrapper.loadServlet(StandardWrapper.java:941)
		at org.apache.catalina.core.StandardWrapper.load(StandardWrapper.java:838)
		at org.apache.catalina.core.StandardContext.loadOnStartup(StandardContext.java:4180)
		at org.apache.catalina.core.StandardContext.startInternal(StandardContext.java:4480)
		at org.apache.catalina.util.LifecycleBase.start(LifecycleBase.java:164)
		at org.apache.catalina.core.ContainerBase.addChildInternal(ContainerBase.java:599)
		at org.apache.catalina.core.ContainerBase.addChild(ContainerBase.java:571)
		at org.apache.catalina.core.StandardHost.addChild(StandardHost.java:603)
		at org.apache.catalina.startup.HostConfig.deployWAR(HostConfig.java:1014)
		at org.apache.catalina.startup.HostConfig$DeployWar.run(HostConfig.java:1866)
		at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
		at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
		at org.apache.tomcat.util.threads.InlineExecutorService.execute(InlineExecutorService.java:75)
		at java.base/java.util.concurrent.AbstractExecutorService.submit(AbstractExecutorService.java:123)
		at org.apache.catalina.startup.HostConfig.deployWARs(HostConfig.java:816)
		at org.apache.catalina.startup.HostConfig.deployApps(HostConfig.java:468)
		at org.apache.catalina.startup.HostConfig.start(HostConfig.java:1584)
		at org.apache.catalina.startup.HostConfig.lifecycleEvent(HostConfig.java:312)
		at org.apache.catalina.util.LifecycleBase.fireLifecycleEvent(LifecycleBase.java:109)
		at org.apache.catalina.util.LifecycleBase.setStateInternal(LifecycleBase.java:385)
		at org.apache.catalina.util.LifecycleBase.setState(LifecycleBase.java:332)
		at org.apache.catalina.core.ContainerBase.startInternal(ContainerBase.java:776)
		at org.apache.catalina.core.StandardHost.startInternal(StandardHost.java:721)
		at org.apache.catalina.util.LifecycleBase.start(LifecycleBase.java:164)
		at org.apache.catalina.core.ContainerBase$StartChild.call(ContainerBase.java:1203)
		at org.apache.catalina.core.ContainerBase$StartChild.call(ContainerBase.java:1193)
		at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
		at org.apache.tomcat.util.threads.InlineExecutorService.execute(InlineExecutorService.java:75)
		at java.base/java.util.concurrent.AbstractExecutorService.submit(AbstractExecutorService.java:145)
		at org.apache.catalina.core.ContainerBase.startInternal(ContainerBase.java:749)
		at org.apache.catalina.core.StandardEngine.startInternal(StandardEngine.java:211)
[... 176 lines cut ...]
mysql> 

```

For instance, the table "users" is clearly missing the columns "hideMenu" and "theme", which are defined in _V2_7_0_1__UserParameters.java_ (line 13):
```
jdbcTmp.execute("ALTER TABLE users " +
                "ADD hideMenu BOOLEAN DEFAULT false, " +
                "ADD theme VARCHAR(255) DEFAULT 'DEFAULT';");
```
The table "events" is missing column "shortMessage" which is defined in _V2_6__.java_ (line 43):

`jdbcTmp.execute("ALTER TABLE events ADD shortMessage LONGTEXT;");`

And so on...

(The latest Docker version creates the database correctly.)

Scada-LTS version: v2.7.8 (Github ref. 6791491)
Tomcat version: 9.0.95
Java version: JRE 17.0.12+7-Debian-2deb12u1 (also tried with jdk-11.0.2)
Mysql version: 8.4.2 MySQL Community Server
OS: Debian 12.7 - Linux kernel 6.1.0-25-amd64

[image]

Limraj, 2024-10-02:
Hi @felipecarneiro,
what is the situation, has the problem been solved?

Regards.

felipecarneiro, 2024-10-03:
Hi,
I didn’t have time to test this week yet, I’m going to replicate the
scenario again and post the test results.
Thanks!

Limraj, 2024-10-03:
Hi @felipecarneiro,
you can ask here:
https://github.com/SCADA-LTS/Scada-LTS/discussions