SCADA-LTS/Scada-LTS #3029: Working with MQTT milliseconds and display interface #3028

Silonei-Duarte, 2024-10-07:
Good afternoon, dear all, I'm new here, and I'm from Brazil.

I've been developing an interface to capture MQTT data in SCADA_LTS, which is updated in milliseconds, since they are critical sensors.

However, the interface, both in the configuration and in the Datapoints display, only displays seconds. I realized that it is possible to collect data in milliseconds by adding a 0 in front, 0500 seconds,
although the interface then changes to 500 seconds, the data is still collected with reference to 0500

However, the interface in general still only displays in seconds; I understand that it goes beyond a configuration file. Can you tell me where to start to implement this change or if there are any plans to add milliseconds to the code in future native versions.

Limraj, 2024-10-09:
Hi @Patrykb0802,
Please note that the issue number is **3029** and not 3028, 

Regards,
Kamil Jarmusik

Limraj, 2024-10-10:
Hi @Silonei-Duarte,
The Update parameter for this Data Source means the period during which the connection for the Data Source is checked and repeated, if necessary.

This will be saved (for int type 0200=200), but the Data Source connection (connection refresh) will be checked within a period of 200 seconds. 

However, this Data Source is designed in such a way that if there is a change in the value on the subscribed topic, it immediately updates the value for the points, regardless of the Update parameter, so you can observe that it works as you expect.

Regards,
Kamil Jarmusik

Limraj, 2024-10-15:
Hi @Silonei-Duarte,
1. Let me know if the description of the MQTT Data Source operation, and specifically the Update parameter, is clearly explained by me?
2. Do you expect to be able to set this parameter (Update) in milliseconds?

I hope I understood you correctly. 

Regards,
Kamil Jarmusik

closed 2024-10-23 by Limraj