SCADA-LTS/Scada-LTS #3068: Scaling factors in SNMP datasources

nduvernois, 2024-12-17:
In Modbus IP datasources, it is possible to define a scaling for the data (y = a * x + b) for numeric datapoints. This is not possible in SNMP datasources, neither in Bacnet IP datasources (and maybe others, I have not checked). 

This is not convenient if the data includes a scaling factor (e.g. in standard SNMP data for UPS RFC1628, Frequency is sent with precision 0.1 Hz, currents with 0.1 A which means that I currently display data in dHz and in dA). I also had once a case with temperature sent with 0.1°C precision, which leads to display "d°C", which is really strange.

The current solution to display in "regular" units is to define a metadata and perform the conversion in this way. Integrating scaling in the datapoint definition would avoid to create a new datapoint, a new storage need, and a risk of confusion.

nduvernois, 2024-12-17:
Sorry, I forgot this : v2.7.7.3_For build 3789793776, Windows Server 2016.

Limraj, 2024-12-18:
Hi @nduvernois,

We have already discussed this topic here (#2359) and it follows that in order to obtain unambiguity between the conversion and the original value, we can introduce separate multiplication or (not and) addition factors. A quadratic equation can have two equivalent solutions, so this destroys this uniqueness. ( [Bijection](https://en.wikipedia.org/wiki/Bijection) ) This conversion from the scaled value to the original value would be necessary if we wanted to send some value to the PLC. If the value were scaled by a quadratic equation, we could have two equivalent and different solutions and it is not known which value should be sent.

Detailed explanation:
"we want to add the proposed converter:
ax^2 + bx + c = y
The situation when we get the value from the driver is simple, we substitute for x and calculate y.
However, in the case of setting the value at such a point, i.e. a situation where we have y and we want to calculate x, the situation becomes a bit more complicated. That is, if the delta of such an equation is greater than 0, it means that the equation has two solutions, that is, we have the equation:

ax ^ 2 + bx + c1 = 0

where:
c1 = c - y;

then:
delta = b ^ 2 - 4 * a * c1

If delta> 0 then there are two equivalent solutions x1 and x2.
How does it look in practice? Is such a situation permissible?"

"
However, if we approached the problem in a different way, praised the raw values, and depending on the context, displayed the converted, then here we could develop a universal mechanism.
"

To display the value itself, we can format the value and add its unit:
[image]

Regards,
Kamil Jarmusik

nduvernois, 2024-12-18:
Hello. I agree with the conclusion of post #2359. I know the solution of formatting the datapoint, but, coming back to this temperature sensor : lets assume it would measure 25.5°C. The number transmitted in the SNMP format was 255. Then how may I display it ? 255 d°C (deci-celsius degree) ? this is really awkward. So I created a metadata point to perform the division by 10, so that I could finally display a regular 25.5°C.