SCADA-LTS/Scada-LTS #3073: Running on Gentoo rv64

moroznah, 2025-01-09:
**Describe the bug**
Trying to run Scada-LTS natively on Gentoo in a not very ugly way.
End goal is to run on rv64 arch, but i'm unable to launch it even on amd64.
Crashes with:  java.lang.NullPointerException: null
Please help

**To Reproduce**
Easy to reproduce, on a Gentoo system run:

```
source /etc/profile
emerge dev-java/openjdk:11 tomcat:9
useradd -m -G users,wheel,realtime -s /bin/bash scadalts
/usr/share/tomcat-9/gentoo/tomcat-instance-manager.bash --create --suffix scadalts --user scadalts --group scadalts
cd /var/lib/tomcat-9-scadalts/webapps
wget https://github.com/SCADA-LTS/Scada-LTS/releases/download/v2.7.8/Scada-LTS.war
mkdir /var/lib/tomcat-9-scadalts/lib
cd /var/lib/tomcat-9-scadalts/lib
wget https://github.com/SCADA-LTS/Scada-LTS/raw/refs/heads/develop/tomcat/lib/activation.jar
wget https://github.com/SCADA-LTS/Scada-LTS/raw/refs/heads/develop/tomcat/lib/jaxb-api-2.4.0-b180830.0359.jar
wget https://github.com/SCADA-LTS/Scada-LTS/raw/refs/heads/develop/tomcat/lib/jaxb-core-3.0.2.jar
wget https://github.com/SCADA-LTS/Scada-LTS/raw/refs/heads/develop/tomcat/lib/jaxb-runtime-2.4.0-b180830.0438.jar
wget https://github.com/SCADA-LTS/Scada-LTS/raw/refs/heads/develop/tomcat/lib/mysql-connector-java-5.1.49.jar
ln -s /var/log/tomcat-9-scadalts /var/lib/tomcat-9-scadalts/logs
echo 'TOMCAT_JVM="openjdk-bin-11"' >> /etc/conf.d/tomcat-9-scadalts
echo 'CATALINA_OPTS="-Xmx2G -Xms2G"'  >> /etc/conf.d/tomcat-9-scadalts

#edit for db credentials /var/lib/tomcat-9-scadalts/conf/context.xml
rc-service tomcat-9-scadalts start
```
Prior to running natively, this db was tested successfully with Scada-LTS amd64 docker image and worked fine.
`docker run --name scadalts -v /etc/scadalts/context.xml:/usr/local/tomcat/conf/context.xml scadalts/scadalts
`

**Additional context**
Attached non-empty log files:
[catalina.2025-01-09.log](https://github.com/user-attachments/files/18360096/catalina.2025-01-09.log)
[flyway.log](https://github.com/user-attachments/files/18360097/flyway.log)
[localhost.2025-01-09.log](https://github.com/user-attachments/files/18360098/localhost.2025-01-09.log)
[mango.log](https://github.com/user-attachments/files/18360099/mango.log)

Limraj, 2025-01-09:
Hi @moroznah,

1. Here is the installer for Linux:
https://github.com/SCADA-LTS/linux-installer
However, you may have problems working on this system, we support family systems based on Ubuntu, with basic programs and system tools installed.

2. In the case of docker, we recommend using docker-compose with the configuration available in the sources:
https://github.com/SCADA-LTS/Scada-LTS/blob/develop/docker-compose.yml
**This method seems to be the safest solution. However, if you want to use our application in production in this way, Docker knowledge is required, including the importance of learning about Volumes, which enable the persistence of container state.**
All images available, we recommend using v2.7.7.3 or v2.7.8:
https://hub.docker.com/r/scadalts/scadalts/tags

3. We support origin MySQL server, not MariaDB. There are differences between these versions of MySQL server and we cannot guarantee that everything will work on the MariaDB server.

Regards,
Kamil Jarmusik

moroznah, 2025-01-14:
@Limraj 
Thanks for answering!

1. Yes i've seen and tested these scripts on amd64. Dissecting them helped a lot to understand how to run Scada-LTS on non-Debianish distros.

2. I agree that this is probably best approach, however there is no riscv64 image for docker. Also i doubt it that you are willing to support a riscv64 image now since this arch is not mainstream yet.

3. To tell you the truth i haven't even tested MySQL. Everything seems to be working fine on MariaDB.

4. CATALINA_HOME was the problem. I've switched to 2.8.0 and it's running on riscv64! Previously posted Gentoo script does the job. I don't use many connectors, mostly Modbus TCP and HTTP retrievers, these work fine. Will test more soon, mostly energy sector related, such as IEC101.

5. I agree with you, this is so true for many distros.

If you don't mind, i will post here if will encounter any specific riscv64 issues. At some point this arch is going to be mainstream and many ppl will start swapping arm64 SBCs. Perhaps this will help to do necessary tweaks for a future docker image.
I'm also doing extensive performance testing to get a good comparison of performance on various arches, compile flags and OpenJDK versions (for ex. Eclipse vs natively compiled).

Regards,
Igor

Limraj, 2025-01-15:
Hi @moroznah,
We are curious about your results and observations with MariaDB and riscv64. 
Generally, we are considering switching to the MariaDB server. 

However, remember that version 2.8.0 is not ready for production yet, we rather recommend using this docker and version v2.7.8 or v2.7.7.3 (instead of using the MySQL server image, you can configure the MariaDB server image similarly), or another recommended installation method that I mentioned. While the problem with CATALINA_HOME is simply we set this variable for the user and it should help, except that we do not use the tomcat installer, which probably creates a system user that does not have its own home directory, which may make this configuration difficult. 

Regards,
Kamil Jarmusik

Limraj, 2025-02-15:
Hi @moroznah,
We are planning release 2.8.0. The current state of this version is incompatible with your version 2.8.0. 

If you want to run the released version 2.8.0, you need to clear the events table first. 

Due to the fact that such incompatibilities may occur while working on a version, we do not recommend using versions that have not been completed, which is why I suggested using other versions.

Regards,
Kamil Jarmusik

moroznah, 2025-02-18:
Hi @Limraj
Thanks about this info. Will update before deploying. Are there more possible changes in db before 2.8.0 release? I'm thinking of deploying in production.
2.7.8 still refuses to work due to CATALINA_HOME. Tried multiple combinations, alsways gives "java.lang.NullPointerException: null"

I'll keep initial post updated on further work.

Have anyone tried PREEMPT_RT kernels? It's mainline since 6.12.
On rv64 RT kernel java dumps:
```
[Feb11 22:37] BUG: sleeping function called from invalid context at kernel/locking/spinlock_rt.c:48
[  +0.000017] in_atomic(): 0, irqs_disabled(): 1, non_block: 0, pid: 2927, name: java
[  +0.000006] preempt_count: 0, expected: 0
[  +0.000004] RCU nest depth: 0, expected: 0
[  +0.000006] CPU: 3 UID: 1001 PID: 2927 Comm: java Not tainted 6.12.13-gentoo-riscv64 #24
[  +0.000007] Hardware name: milkv Milk-V Mars CM/Milk-V Mars CM, BIOS 2025.01-00001-gced64a332cea 01/01/2025
[  +0.000005] Call Trace:
[  +0.000004] [<ffffffff80006066>] dump_backtrace+0x26/0x3c
[  +0.000015] [<ffffffff80871d7e>] show_stack+0x2c/0x3e
[  +0.000010] [<ffffffff808802c0>] dump_stack_lvl+0x52/0x7a
[  +0.000010] [<ffffffff808802fc>] dump_stack+0x14/0x1e
[  +0.000008] [<ffffffff8004d396>] __might_resched+0x14a/0x154
[  +0.000010] [<ffffffff8088a0ea>] rt_spin_lock+0x2c/0x66
[  +0.000008] [<ffffffff800320dc>] force_sig_info_to_task+0x26/0x128
[  +0.000008] [<ffffffff800327a8>] force_sig_fault+0x56/0x82
[  +0.000007] [<ffffffff80005c52>] do_trap+0x28/0xbe
[  +0.000007] [<ffffffff80880734>] do_trap_insn_illegal+0xde/0x100
[  +0.000008] [<ffffffff8088c690>] handle_exception+0x150/0x15c
```

Limraj, 2025-02-18:
Hi @moroznah,
As I have already mentioned, you need to set the CATALINA_HOME system variable for the user who runs the tomcat process. To achieve this, you probably cannot use a user created by an automatic tomcat installer downloaded from the Internet, because it is probably a service (or system) type user, it does not have its own home directory, and not default shell, and by default you cannot log in to such a user because it has no password set, of course you can play around, set the password hash, set shell, then use the export command, etc. But I don't know if it makes sense. We therefore advise against such installation. 

We recommend using Docker or our installer, which I linked in previous messages. 

If users use released versions, we provide automatic migrations and there should be no problem with a newer version. 
Of course, there is a risk that if we switch from an old version to a much newer one, some problems and difficulties may arise, but they should be solved outside the code. 

**It should be mentioned, however, that if a migration occurs in a given version and the database structure is changed, such a change cannot be easily restored without a backup of the old database.  Therefore, we recommend making such backups, especially before installing a new version.**

In practice, we do not roll back the application version. I don't remember this happening to us. However, this is possible if we have a backup of the old database.

Regards,
Kamil Jarmusik

moroznah, 2025-02-18:
@Limraj 
There is a real scadalts user and group, with a home folder and .bashrc. Changing CATALINA_HOME in .bashrc does not solve anything, crashes with same "java.lang.NullPointerException: null" errors.
I'm not concerned about this since it's fixed in 2.8.0.

There is no Docker for rv64.

OpenJDK works fine on rv64. It's also supported on my distro:
https://packages.gentoo.org/packages/dev-java/openjdk
Previously provided dump was from RT kernel. It would be interestng to run and benchmark on Real-Time system.

I'm already in contact with Brent.

Regards,
Igor

Limraj, 2025-02-18:
Hi @moroznah,
How do you run tomcat from Scada-LTS. Do you log in as this user, check whether the CATALINA_HOME system variable is set and run Tomcat from Scada-LTS directly from it? 

If you do it differently, there is no certainty that you are actually running Tomcat where the CATALINA_HOME system variable is available in the system. Let me emphasize once again that this is not the same as launching a process with the CATALINA_HOME parameter set. This variable must be available outside the tomcat process.

Also, have you used our installers? 

Regards,
Kamil Jarmusik

Limraj, 2025-02-18:
@moroznah So we close this issue?

moroznah, 2025-02-18:
@Limraj 
Oh i see what you ment now about trying to start by hand Tomcat with CATALINA_HOME under specific user (without any scripted wrappers). Will try this and see where it leads.

Your installers do the job fine on x86. From them i also got basic understanding how to run Scada-LTS natively.

Regarding crash dump backtrace, i think you misunderstood my post.
OpenJDK is stable on rv64, at least for me and all the software that i have tested.
That specific backtrace probably has nothing to do with rv64 specifically, but with PREEMPT_RT kernel patches.
https://wiki.linuxfoundation.org/realtime/start
OpenJDK will probably crash on x86 also with RT enabled.
I'm just curious if anyone has running Scada-LTS on Real-Time kernel. The main benefit would be decreased latencies. This still would not be comparable for ex. with OpenPLC, but opens possibilities to control timing critical tasks.

Regards,
Igor

moroznah, 2025-02-18:
> [@moroznah](https://github.com/moroznah) So we close this issue?

If you don't want any updates about rv64, sure you can close it, it's mostly solved.

Regards,
Igor

Limraj, 2025-02-18:
> [@Limraj](https://github.com/Limraj) Oh i see what you ment now about trying to start by hand Tomcat with CATALINA_HOME under specific user (without any scripted wrappers). Will try this and see where it leads.

This is about making sure of the setup. It's also best to set ownership for tomcat policies.

Limraj, 2025-02-18:
However, since you have contact with Brent, we'll close the topic.

You can create a discussion here if you have something to say about this topic:
https://github.com/SCADA-LTS/Scada-LTS/discussions

Issue is used to indicate problems.

closed 2025-02-18 by Limraj

Limraj, 2025-02-19:
@moroznah 
First, check in the console whether you actually have the system variable set:
`echo $CATALINA_HOME`

(printenv for all variables)

And make sure that after changing the .bashrc file, you used the command that loaded the changes:
`source .bashrc`

If I say that if a particular problem is caused by something specific, then it most likely is.