SCADA-LTS/Scada-LTS #3093: Error in ScadaLTS

Lclaros1985, 2025-02-18:
error in ScadaLTS when trying to access via url outside the local host, a link redirected to localhost:8085/Scada-LTS was defined, application opens, tomcat ok, however the graphical representation, does not visualize even if nothing is published to view external access, in local everything ok, everything released in firewall, ports, observed error presented in MiscDwr mime content type

[image]

Limraj, 2025-02-19:
Hi @Lclaros1985,
The fastest way is probably to use Firefox. 

This is a browser limitation, it may also be a result of setting the Content-Security-Policy header, with a specific parameter: https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Content-Security-Policy

Our application sends these js files with the MIME type text/plain, and the browser does not want to execute scripts in files that have this MIME type instead of application/javascript. 
Unfortunately, we did not experience this problem locally or on our servers, so it may also be the result of restrictive security settings in your browser, or it may concern a specific version of Chrome. I don't know how this will work when JS scripts are in HTML files, because such files are also not returned with the application/javascript MIME type, so I don't know to what extent this setting is applicable in practice.

This is something you might want to look into to set the correct MIME types for your js script files. 
https://stackoverflow.com/questions/39228657/disable-chrome-strict-mime-type-checking

Regards,
Kamil Jarmusik

Lclaros1985, 2025-02-19:
Thank you very much kamil, I will try to use Firefox, the problem will be using it in the mobile browser

Lclaros1985, 2025-04-01:
yes , and it didn't work, the way I found it was to use the reverse proxy oh it's working

Limraj, 2026-01-08:
Hi @Lclaros1985,
If you need closer collaboration, technical support, or the addition of a new feature, we encourage commercial cooperation:[scada-lts@softq.pl](mailto:scada-lts@softq.pl)

Best regards.