SCADA-LTS/Scada-LTS #3107: HTTP receiver listening trouble

nex2000, 2025-03-17:
Hello , I am testing the listener functionality in scadaLTS. Unfortunately I can't get it to work on scadaBR but not on LTS. I did a new installation of version 2.7.8 but I get 403 and LTS stays in listening without receiving anything. I also tried with Postman and basic auth passing the parameters but nothing happens. I tried on windows and Linux with the same behavior. I send this command

http://127.0.0.1:9700/Scada-LTS/httpds?data=0

I read that there were problems on this topic there is a solution. Thanks

[image]

nex2000, 2025-03-17:
Now I tested the 2.6.18 version It's work fine , the trouble is in te updated version.

regards

Limraj, 2025-03-17:
Hi @nex2000,
We have increased the security level, it works as before but only via https, or via http, but after enabling a special user, details here:
https://github.com/SCADA-LTS/Scada-LTS/wiki/Configuration-HTTP-Receiver-Data-Source-%5Bhttpds-API%5D

Regards,
Kamil Jarmusik

nex2000, 2025-03-18:
Thanks a lot

nex2000, 2025-03-18:
Hello I tested it as your suggestion , I receive status cose 200 but nothing listen

[image]

[image]

[image]

Limraj, 2025-03-18:
Hi @nex2000 ,

As I wrote and it is in the wiki:
1. You need to configure HTTPS communication for tomcat:
https://github.com/SCADA-LTS/Scada-LTS/wiki/Configuration-of-HTTPS-communication-for-Tomcat

2. Either we use a special user httpds-basic, which must be enabled, by default it is disabled, the default password is httpds-basic, we use this user in basic authentication, then you can operate on http, then https is not necessary.

Regards,
Kamil Jarmusik

nex2000, 2025-03-18:
hello , sorry for ma trouble
 I also tried with user https-basic , i got status code 200 (accepted) but not show nothing in the box listen.
I use 2.7.8 release

nex2000, 2025-03-18:
sorry , I made a mistake, sorry, now it works fine thanks a lot

Limraj, 2025-03-19:
Hi @nex2000,
I'm glad you managed to configure it and it works.

Regards,
Kamil Jarmusik

closed 2025-03-19 by Limraj