SCADA-LTS/Scada-LTS #3115: Time Events

Lclaros1985, 2025-03-25:
I needed help if the functionality already exists. Because in the old ScadaBR 1.0 this works very well, I have a certain point it has upper and lower level status alarms, no change, anyway in 1.0 the alarm keeps "playing the sound" until it is recognized, in Scada-LTS this does not happen it "plays the sound only once", then if the user does not notice it ends up forgetting, would there be some configuration to be done or would it be a failure I await friends

Limraj, 2025-03-26:
Hi @Lclaros1985,
there is a problem that introduces the possibility of configuring this sound:
https://github.com/SCADA-LTS/Scada-LTS/issues/2657

Currently there is one signal, because it did not work for a long time and we did not want to introduce a sound revolution for customers.

Regards,
Kamil Jarmusik

Lclaros1985, 2025-03-26:
Hi Kamil, forgive me because I didn't understand your answer, wouldn't it be possible to leave the alarm sounding for longer, or would it be possible for me to customize this for my project?