SCADA-LTS/Scada-LTS #3116: How to disable new UI in last version

nex2000, 2025-03-29:
hello I would like to use the latest version 2.7.8 but disable the new UI, unfortunately some users get confused with the two interfaces, they are used to the old interface. I can disable the new UI and stay in 2.7.8. Thanks

Limraj, 2025-03-29:
Hi @nex2000,
I think we will discuss this as a team.

Regards,
Kamil Jarmusik

nex2000, 2025-03-30:
Thanks for your help

Regards

Limraj, 2025-04-10:
Hi @nex2000,

1. How do users get to the new UI? Is it the popup like in this issue? #3124 Or maybe it's about the functions in System Settings?  Some settings are only possible on the new UI and are redirected to it. #3102
3. How would you imagine disabling the new UI?